python-control/python-control #48: lsim is slow

cwrowley, 2015-03-20:
The `lsim` command can be much, much slower than `scipy.signal.lsim`.  For instance, consider the following example (run in IPython for `%timeit`):
```python
import scipy.signal as sig
import control
import numpy as np
t = np.arange(0,1000,0.5)
u = np.sin(2.*t)
sys1 = sig.lti(1.,[1,0,1])
sys2 = control.tf(1.,[1,0,1])
%timeit sig.lsim(sys1,u,t)
%timeit control.lsim(sys2,u,t)
```
For me, `scipy.signal.lsim` gives 17.4 ms per loop, and `control.lsim` gives 7.64 s per loop.

The current implementation of `control.lsim` calls a general-purpose ODE solver at each step.  I think it is better to assume a linear system with constant-spaced timesteps in `t` (as done in `scipy.signal.lsim` and in Matlab's `lsim`), and then one can evaluate the matrix exponential once and apply it at each step.  If non-uniform timesteps are desired (which is very rare, at least in my experience), then we could have an alternative routine `control.lsim2` that calls the general-purpose ODE solver--that is what is done in `scipy.signal`.

murrayrm, 2015-03-21:
Since I think we already depend on `scipy.signal`, can we just call through to that?

cwrowley, 2015-03-21:
Unfortunately, no, because the LTI objects in `scipy.signal` are totally different from the `TransferFunction` and `StateSpace` objects in `control`.

This raises another issue, that ideally it would be nice if these objects were all compatible with one another, since a lot of stuff is already included in `scipy.signal`.  However, those are pretty different, so that would require a lot of work.

blinkminster, 2015-03-21:
Defaulting to the general-purpose solver was also motivated by how scipy's
lsim doesn't work for systems with a pole at the origin, which is
relatively common in control systems. So any function that tries to use
scipy.signal.lsim or whatever algorithm that uses should also fall back to
the general-purpose solver if there is a zero at the origin.

Sawyer

------------------------
Sawyer B. Fuller, Ph.D.
Harvard School of Engineering and Applied Science
Wyss Institute for Biologically Inspired Engineering
http://people.seas.harvard.edu/~minster/

cwrowley, 2015-03-21:
I was just planning to call `scipy.linalg.expm(A * dt)` where `A` is from the state-space representation of the system and `dt` is the (constant) timestep.  I believe that should work fine even if there is a zero at the origin.

blinkminster, 2015-03-21:
I've never looked into why scipy's lsim fails with a pole at the origin,
but if what you're talking about is faster than a standard ode solver, is
correct, and works for a zero pole, then that sounds great to me.

Sawyer

------------------------
Sawyer B. Fuller, Ph.D.
Harvard School of Engineering and Applied Science
Wyss Institute for Biologically Inspired Engineering
http://people.seas.harvard.edu/~minster/

PR #51 (2015-03-29): Speed up forced_response/lsim
Fixes #48.

This version of lsim assumes a constant timestep, as in Matlab.  This algorithm is much faster than the previous implementation, which used a generic ODE integrator: the example in issue #48 now runs over 400x faster, and is now a little faster than scipy.signal.lsim.  This algorithm is more general than the scipy version, since it works even if the system has a pole at the origin.

If we want a version that works with variable timesteps, I suggest adding an `lsim2` function, as in `scipy.signal`, which could call the generic ODE integrator as before.  [... cut]
commit d7d278ba (2015-03-29): Speed up forced_response (lsim)

Fixes #48

The old algorithm called a general-purpose ODE integrator at each
timestep, and this could be extremely slow (hundreds of times slower
than scipy.signal.lsim).  This new algorithm evaluates the matrix
exponential once, and uses it to advance the simulation at each step.

Importantly, the old algorithm would work for variable timesteps, but
the new version assumes a fixed timestep (as does Matlab's lsim).  This
is the usual case, and it is the price one pays for increased speed.

Note that this algorithm is the same idea as the algorithm used in
scipy [... cut]
closed 2015-03-29 by cwrowley