python-control/python-control #49: Bode, Nyquist, and root locus plots are crude

cwrowley, 2015-03-20:
Bode, Nyquist, and root locus plots are all much nicer in Matlab, both because they make smarter choices for things like frequencies to plot, and because their plots are interactive, so you can click on points to get additional information, etc.
I have some ideas for how to improve this (e.g., more sophisticated ways of generating frequencies for Bode/Nyquist plots, with finer resolution near lightly damped poles or zeros; better gains for root locus plots, with finer resolution near break-out and break-in points, etc), and I'm planning to try implementing those.

slivingston, 2015-03-20:
Regarding making plots interactive, you could use events from Matplotlib to respond to clicks, e.g., using [matplotlib.backend_bases.MouseEvent](http://matplotlib.org/api/backend_bases_api.html#matplotlib.backend_bases.MouseEvent).

roryyorke, 2015-06-16:
My 2c: don't forget about Nichols.  It might be worth separating the frequency response and root-locus into separate issues.

For the frequency plots, one could also consider showing low and high frequency asymptotes.
 
One idea I had for this to have the number of sample points be a multiple of the number of poles and zeros, though that is a bit simplistic.  Another idea is to relate the density of the sample points to the curvature of the graph (though I don't know how to turn that into an algorithm...)

repagh, 2015-07-18:
one thing we also need are damping-omega grids for root-locus. 

for root-locus, having the break-in and break-out points in the gain vector already helps,

Sup3rGeo, 2017-10-25:
Concerning the damping-omega grids as @repagh said, I have written a simple function that draws the grid for Z-domain (zgrid):

https://gist.github.com/Sup3rGeo/17d0eb2b0f4d60c1ebd24695ac692b5d

Is there any possibility for this to be added to the python-control functions? I have never contributed to open source projects before, but if you tell me if what to do I can perhaps create a pull request for this.

Sup3rGeo, 2018-02-19:
Created a PR for this (#193)

don4get, 2019-01-05:
Can we have more of what might be improved from what is already present in the library?

Thanks to all the improvements you done since the creation of this issue, I can't say if it's closed or not.

rxa254, 2020-06-25:
For the Nyquist plot, I would like to see:

1. Smoother lines. This can be done by sampling the system better in regions where |d(Mag)/df| is large.
1. Better scaling (not sure how to do this). With systems that have a lot of DC gain or if there is a deep bandstop filter, its difficult to see the detail just due to the dynamic range.
1. It would be good to have arrows denoting the direction of increasing frequency, as matlab has.
1. This is an example where the frequency is encoded as color changes in the curve (https://lpsa.swarthmore.edu/Nyquist/NyquistExamples.html).

This paper (https://doi.org/10.1109/ECC.2014.6862243) shows a way to plot the Nyquist logarithmically.

murrayrm, 2020-06-25:
Great set of suggestions @rxa254.  If anyone wants to sign up for these changes, that would be great.  Maybe some summer coding practice for @cwrowley?

murrayrm, 2021-02-11:
PRs #522 and #534 address some (but not all) of the suggestions here.  In particular, #522 adds more points for frequency response plots and #534 has directional arrows and other features.

murrayrm, 2021-06-06:
There is some nice looking code for [JuliaControl](https://github.com/JuliaControl/ControlSystems.jl/pull/532#pullrequestreview-673973225) that we might take a look at for improving frequency response plots.