python-control/python-control #50: Interactive plots in IPython notebooks

cwrowley, 2015-03-20:
I find IPython notebooks a great way to work interactively, and it would be nice if inline plots could be interactive (e.g., letting you zoom, or having tool tips show more information).  With matplotlib, though, inline plots are just images, so they cannot be interactive.

There's a new plotting library called [Bokeh](http://bokeh.pydata.org), an open-source project led by Continuum Analytics (the wonderful folks that make [Anaconda](http://continuum.io/downloads)), that looks really nice for making interactive plots like this.  I'd like to try making an alternate backend one could specify with, say, `control.set_backend("bokeh")` that would make commands like `bode()` and `nyquist()` generate plots using Bokeh.  The default could still be matplotlib, of course.

Any thoughts or suggestions for other libraries that might be useful for making interactive plots?

slivingston, 2015-03-20:
Outside of IPython notebooks, you can zoom-in, pan, etc. in Matplotlib plots. I just shared a link via issue #49 to [matplotlib.backend_bases.MouseEvent](http://matplotlib.org/api/backend_bases_api.html#matplotlib.backend_bases.MouseEvent) for responding to clicks on points in these plots.

Perhaps there can be two major interactive use-cases: with and without IPython. The former case could be treated with Bokeh, and the latter case with Matplotlib and various GUI hooks that we create.

blinkminster, 2015-03-21:
Clancy,

I use the following code in my ipython notebooks and it makes the plots
interactive (zooming, panning work; this is in the standard Anaconda
distribution on OSX 10.10):

import matplotlib as mlt
mlt.use('nbagg') # makes plots interactive for zooming.

I haven't tested it with python-control yet but I don't see why there would
be a problem.

- Sawyer

------------------------
Sawyer B. Fuller, Ph.D.
Harvard School of Engineering and Applied Science
Wyss Institute for Biologically Inspired Engineering
http://people.seas.harvard.edu/~minster/

cwrowley, 2015-03-21:
@blinkminster, that is pretty slick!  I never new about nbagg.  (Incidentally, this also apparently works with the IPython magic `%matplotlib nbagg`.)  I still might give the Bokeh library a try, but this is great to know about.

ilayn, 2017-01-02:
I am also wondering about this issue for my library and I have been also trying to find a good basis for this issue. So far I've been experimenting with Bokeh. Things are still wonky as an experiment but I just wanted to also get your opinion about the following prototype

http://nbviewer.jupyter.org/gist/ilayn/f8cf8b0a68c6839dbcc346884511e2ed

There are some issues of course since I'm still learning JS magic but I guess it might serve as a basis. Though I don't follow the same structure as python-control the data will come from somewhere anyway for frequency responses. Hence, from the point on where one has the complex number array this can be used independently. Maybe python-control devs can use this and put more momentum to this kind of development?

murrayrm, 2018-12-28:
See PR #204 and #209 for some initial implementations of this type of functionality.

lucasbellinaso, 2021-01-13:
Hello,
I teach control systems.  Due to the pandemic, students not accessing the University, I created an interactive interface with Bokeh for using in Google Colab similar to Sisotool (Matlab).
[https://github.com/lucasbellinaso/PythonSisoDesignApp](url).