python-control/python-control #57: Mistakes in examples in documentation

cwrowley, 2015-04-09:
I noticed a minor mistake in the documentation for `_convertToTransferFunction`.  In particular, in the example
```
sys = _convertToTransferFunction([[1. 0.], [2. 3.]])
```
the numbers should be separated by commas.  That made me wonder if there were mistakes in other examples, so I ran all of the examples as doctests with `nosetests --with-doctest`.  That reveals 38 failed tests, and 1 error.  Many of these problems are because some quantities are not defined in the examples.  For instance:
```
T, yout = initial_response(sys, T, X0)
```
gives `NameError: name 'sys' is not defined`.  I think it would be good to make these examples all self-contained and "runnable", so that 1) they are complete (e.g., users know exactly how to define an object like `sys` in the example above); and 2) they can be automatically tested as doctests.  Any comments or objections?

murrayrm, 2015-04-09:
My only concern would be that some examples might require a lot of setup that would make it verbose and less useful.  Setting up something like an LQR problem, for example.  But if we can do these succinctly, then I agree that we should have the goal that examples execute (so `nosetests --with-doctest` returns no errors).

cwrowley, 2015-04-09:
I agree about the potentially verbose setup.  LQR is a good example too, because the calling syntax is complex, and not immediately obvious from the function's signature `lqr(*args, **kwargs)`.  However, in the [LQR documentation](http://python-control.readthedocs.org/en/latest/synthesis.html), I think the "Examples" section could actually be removed, since the syntax is already illustrated earlier in the doc string.  Maybe formatting something like this could work for other routines where the setup is complicated too.

murrayrm, 2018-12-23:
Fixed in PR #228.

closed 2018-12-23 by murrayrm