python-control/python-control #61: Stability Margin may give incorrect results with FRD system

callawaycass, 2015-04-22:
When calling stability_margins() with an FRD system, or with a (mag, phase, omega) system that is converted to FRD, the phase crossover frequency in particular (and consequently the gain margin) can be radically inaccurate. Sometimes I even get the phase crossover as a negative frequency! I am working with a fairly simply system with a zero, a couple poles, no resonances, etc., so I looked to see if I could figure out what was going on.

I found that changing lines 207 and 208 in margins.py from:

    wc = np.array([sp.optimize.fsolve(mod, sys.omega[0])])[0]
    w_180 = np.array([sp.optimize.fsolve(arg, sys.omega[0])])[0]

to:

    wc = np.array([sp.optimize.fsolve(mod, np.median(sys.omega))])[0]
    w_180 = np.array([sp.optimize.fsolve(arg, np.median(sys.omega))])[0]

Always produces the expected result. It makes sense to seed the fsolve function with the median frequency in the defined FRD system because one would expect the crossovers to generally be in "the middle" of the bode plot instead of at the lowest frequency.

Just an improvement I'm using myself, but it may be worth incorporating.

cwrowley, 2015-04-22:
Thanks for pointing this out, and for the proposed fix!  Would you mind providing a small example that reproduces the problem, especially for phase crossover being reported as negative?

callawaycass, 2015-04-22:
I don't know the best way to do this, but here goes... I can't find the results where I was getting negative crossover, but here is an example that gives an incorrect result. I have this data in a tab-delimited file and I'm reading it in using csv.reader. Units are: Frequency (Hz), Phase (degrees), Gain (dB).

Frequency  	Phase      	Gain       
100         	103.0596124 	43.06807989 
109.6478196 	100.9411851 	42.37155698 
120.2264435 	98.85676169 	41.63518924 
131.8256739 	96.83932633 	40.86066824 
144.5439771 	94.91939983 	40.05043469 
158.4893192 	93.12429171 	39.20762079 
173.7800829 	91.47734077 	38.33597267 
190.5460718 	89.99718051 	37.43974492 
208.9296131 	88.69710208 	36.52356256 
229.0867653 	87.58460986 	35.59225128 
251.1886432 	86.66126281 	34.65064515 
275.4228703 	85.92286595 	33.70338705 
301.995172  	85.36002671 	32.75474152 
331.1311215 	84.95903335 	31.80843951 
363.0780548 	84.70296207 	30.86757002 
398.1071706 	84.57288874 	29.93452656 
436.5158322 	84.54907594 	29.01100785 
478.6300923 	84.61202512 	28.09806535 
524.8074602 	84.74331839 	27.19618483 
575.4399373 	84.92621557 	26.30538791 
630.9573445 	85.14600904 	25.42533996 
691.8309709 	85.39016601 	24.55545369 
758.577575  	85.64830295 	23.69498098 
831.7637711 	85.91204143 	22.84308923 
912.0108394 	86.17479059 	21.99892097 
1000        	86.43149378 	21.16163764 
1096.478196 	86.67836708 	20.33044954 
1202.264435 	86.91264832 	19.50463444 
1318.256739 	87.1323675  	18.68354726 
1445.439771 	87.33614389 	17.8666232  
1584.893192 	87.52301123 	17.05337596 
1737.800829 	87.69227026 	16.2433926  
1905.460718 	87.84336693 	15.4363261  
2089.296131 	87.97579429 	14.63188622 
2290.867653 	88.08901697 	13.82982947 
2511.886432 	88.1824171  	13.02994854 
2754.228703 	88.25526157 	12.23206178 
3019.95172  	88.30668969 	11.43600352 
3311.311215 	88.33571985 	10.64161579 
3630.780548 	88.34127168 	9.848742581 
3981.071706 	88.32219777 	9.057227176 
4365.158322 	88.27731684 	8.266913195 
4786.300923 	88.20543824 	7.477649292 
5248.074602 	88.10536811 	6.689296868 
5754.399373 	87.97588939 	5.901739583 
6309.573445 	87.81571164 	5.114893007 
6918.309709 	87.62339133 	4.328712685 
7585.77575  	87.39722755 	3.54319908  
8317.637711 	87.13514081 	2.758398335 
9120.108394 	86.83454379 	1.97439828  
10000       	86.49221247 	1.191319551 
10964.78196 	86.10416535 	0.409301937 
12022.64435 	85.66555743 	-0.37151378 
13182.56739 	85.1705959  	-1.15100814 
14454.39771 	84.61248508 	-1.92910903 
15848.93192 	83.98340901 	-2.70582044 
17378.00829 	83.27455988 	-3.48125268 
19054.60718 	82.47621855 	-4.2556517  
20892.96131 	81.57788786 	-5.02942492 
22908.67653 	80.56847148 	-5.80316029 
25118.86432 	79.43648092 	-6.57763583 
27542.28703 	78.17024372 	-7.35381804 
30199.5172  	76.75808096 	-8.13284953 
33113.11215 	75.18842394 	-8.91602866 
36307.80548 	73.44985008 	-9.70478597 
39810.71706 	71.5310346  	-10.5006634 
43651.58322 	69.42063348 	-11.3053023 
47863.00923 	67.10712915 	-12.1204434 
52480.74602 	64.57867868 	-12.9479428 
57543.99373 	61.82300494 	-13.7897991 
63095.73445 	58.82736381 	-14.648193  
69183.09709 	55.57860879 	-15.5255308 
75857.7575  	52.0633624  	-16.4244883 
83176.37711 	48.26829145 	-17.3480502 
91201.08394 	44.18047418 	-18.2995391 
100000      	39.78783823 	-19.2826308 
109647.8196 	35.07964153 	-20.3013523 
120226.4435 	30.04696098 	-21.3600614 
131825.6739 	24.68314927 	-22.4634063 
144543.9771 	18.98421873 	-23.6162696 
158489.3192 	12.94911698 	-24.8236988 
173780.0829 	6.579871835 	-26.0908316 
190546.0718 	-0.11839618 	-27.4228233 
208929.6131 	-7.13757058 	-28.8247845 
229086.7653 	-14.4666996 	-30.3017333 
251188.6432 	-22.0921622 	-31.8585632 
275422.8703 	-29.9977838 	-33.5000241 
301995.172  	-38.1648652 	-35.2307055 
331131.1215 	-46.5721209 	-37.0550127 
363078.0548 	-55.1955628 	-38.9771244 
398107.1706 	-64.0083973 	-41.000923  
436515.8322 	-72.9810243 	-43.1298932 
478630.0923 	-82.0812319 	-45.366991  
524807.4602 	-91.274666  	-47.7144845 
575439.9373 	-100.525631 	-50.1737739 
630957.3445 	-109.798257 	-52.7451928 
691830.9709 	-119.058023 	-55.427788  
758577.575  	-128.273639 	-58.2190684 
831763.7711 	-137.419195 	-61.1146994 
912010.8394 	-146.476493 	-64.1081069 
1000000     	-155.437328 	-67.189935  

So I convert the system as follows, noting that I need to add 180 degrees to the phase because the raw data is actually from a measured closed loop gain, which includes the negative feedback summing junction:

    sys = control.frdata.FRD(10 ** (gain / 20) * np.exp(1j * (phase + 180) * np.pi / 180), freq, smooth=True)

Then call stability_margins() and get the following:

    >>> control.stability_margins(sys)
C:\Python34\lib\site-packages\scipy\optimize\minpack.py:237: RuntimeWarning: The iteration is not making good progress, as measured by the 
  improvement from the last ten iterations.
  warnings.warn(msg, RuntimeWarning)
    >>> Out[8]: 
    (0.03453522430472155,
    85.88099319950949,
    0.91495418706639464,
    426.84141880874506,
    11506.88822765755,
    77515.758056640625)

So the phase crossover is incorrectly identified as 426.8 Hz, resulting in a gain margin of 0.0345 or -29 dB. In reality, plotting the system shows that the phase doesn't crossover until ~190 kHz, giving a gain margin of 23.4 (or 27 dB).

By the way, this all comes from what may be an unorthodox source for working with the control toolbox. I am using SIMetrix to run Monte-Carlo simulations of a system loop gain, then post-processing the data in order to calculate the gain and phase margins for each run, then perform statistics on the results. So I'm not designing or simulating a system using the control toolbox, but rather post-processing simulation data. I've also used it to post-process data taken from actual system measurements in the lab, but that's a different story.

endolith, 2016-04-30:
Is this the same issue?

```
L = TransferFunction([1e15], [1e1, 1.01e7, 1e11])
mag, phase, omega = bode(L, omega=2*pi*logspace(log10(10**2), log10(10**8), 500), dB=True, Hz=True)
gm, pm, sm, wg, wp, ws = stability_margins((mag, phase, omega))
```

wg should be "Gain margin crossover frequency (where phase crosses -180 degrees)".  From the graph, it should be around 100 MHz, but wg is 1.58 MHz.

wp should be "Phase margin crossover frequency (where gain crosses 0 dB)".  From the graph, this should be around 1.5 MHz, and wp is 1.498 MHz, so that looks right.

(Also, `gm, pm, sm, wg, wp, ws = stability_margins(L)` gives different results, with gm and wg == None.  I don't see anything in the docstring about why this would happen.)

murrayrm, 2017-12-30:
Related to #87