python-control/python-control #65: Separation of plotting from computation

moorepants, 2015-06-15:
If a system is passed in, both the root locus and bode functions do both computations and produce a plot. In my opinion, it would be better to separate the computation from the plotting. The plotting commands should then return matplotlib axes and maybe figure instances and the computation methods would return the results of the computations, e.g. roots, mag/phase, etc.

Right now there aren't clean ways to get hold of the figure and axes (I don't consider `.gca()` and `.gcf()` that "clean"). And if you are using a plotting library other than matplotlib you may not want to call functions that create matplotlib plots.

There maybe other plotting functions in the library that I haven't play with yet too that this is applicable too.

billtubbs, 2021-08-09:
I was thinking along the same lines.  This would be more in line with other libraries like Pandas.  I wrote a draft for how this might work here:

- [New plotting paradigm](https://github.com/billtubbs/python-control-examples/blob/main/plotting-paradigm.md)

Is this what you had in mind @moorepants ?

murrayrm, 2023-12-31:
This issue has been moved to #645.

closed 2023-12-31 by murrayrm