python-control/python-control #67: feedback function

marcino239, 2015-06-26:
Hi,

Just wanted to check feedback function as I couldn't reconcile its results with Matlab.  File xferfcn.py, in function feedback, line  638 is at the moment:

        num = polymul(num1, den2)

But I think it should be:

        num = polymul(num1, num2)

Let me know if this is correct, and I'll send a pull request.

Thanks,
marcino239

murrayrm, 2015-06-26:
Definitely looks like a bug!  When you do the pull request, can you make sure to include a unit test that captures the problem (so the initial code should fail the unit test and the updated code should pass it).

-richard

marcino239, 2015-07-02:
Looks like quite a few test fail if this function is corrected.  The failures are listed below.  It will take me a while to go over this from 1st principles (it's been a while since I've used control theory).

- "control/tests/rlocus_test.py", line 38, in testRootLocus
- "control/tests/frd_test.py", line 129, in testbdalg
- "control/tests/bdalg_test.py", line 174, in testTFTF
- "control/tests/bdalg_test.py", line 157, in testTFScalar
- "control/tests/bdalg_test.py", line 67, in testScalarTF
- "control/tests/bdalg_test.py", line 35, in testScalarScalar
- "control/tests/xferfcn_test.py", line 421, in testFeedbackSISO
- "control/tests/xferfcn_test.py", line 423, in testFeedbackSISO

closed 2015-07-03 by cwrowley