python-control/python-control #71: Documentation on readthedocs.org does not list functions

murrayrm, 2015-10-18:
In the function reference on http://python-control.readthedocs.org/en/latest/control.html, the functions are not actually listed.  Just the headings show up.

whiplash01, 2015-10-18:
Yes, I was earlier looking for them, but except for a permalink there's nothing.

cwrowley, 2015-10-19:
This is actually an issue with sphinx 1.3.1, which unfortunately is the version running on readthedocs now.  I reported this issue in rtfd/readthedocs.org#1290 and this is the same issue described in sphinx-doc/sphinx#1891.  The patch given in sphinx-doc/sphinx#1892 fixes it, but apparently this patch has not been applied to the version used on readthedocs.org.

The python-control docs build fine locally (as long as you use any version of sphinx besides 1.3.1, or v1.3.1 with the patch referenced above), but there doesn't seem to be any workaround for readthedocs besides generating the summary tables manually.  This has been broken since May, by the way, ever since readthedocs.org updated sphinx to 1.3.1 (on 2015-05-04).

So I can think of a few options:
1. Ignore it for now, and wait for sphinx to be updated on readthedocs
2. Generate the tables manually in the python-control documentation (not using autosummary--tedious and not as maintainable, but workable)
3. Host the "official" docs somewhere else, at least until readthedocs is updated

whiplash01, 2015-10-19:
I think we should go with the third option, We should host the "official" docs somewhere else until readthedocs is updated.

slivingston, 2015-10-19:
I recommend the second option, to avoid confusion from multiple and time-varying hosting locations.

murrayrm, 2015-10-24:
Another possibility is to set something up at python-control.org/manual and have this temporarily point to a static copy of the documentation (hosted at Caltech or somewhere else) until readthedocs is finished.  The redirect the URL.  This is related to issue #34, regarding where the home for python-control should be.

cwrowley, 2015-11-04:
Okay, I found a workaround that gets the autosummary tables working on readthedocs, and just submitted PR #72.  I think it's still a good idea to make python-control.org/manual point to http://python-control.readthedocs.org/ (or perhaps even make the main URL python-control.org point there), but we shouldn't need a static copy now (once the PR is merged in).

closed 2015-11-09 by murrayrm

murrayrm, 2015-11-09:
Great fix, Clancy.  I’ve updated the links on python-control.org <http://python-control.org/> to point to the proper place.

What do we need to do to push the updated manual out?  The latest release version is 0.7.0.  The master branch has changes (I think).  Perhaps we create a 0.7.1 release and then use that to push out the manual update?

-richard

cwrowley, 2015-11-09:
The updated manual should now be available at http://python-control.readthedocs.org/en/latest/, but I agree that it would be good to create a 0.7.1 release, since the 0.7.0 release on readthedocs will still have most of the documentation missing.  I will plan to do that in the next few days, and make 0.7.1 available on PyPI and conda/binstar.

PR #72 (2015-11-09): Fix autosummary tables on readthedocs
Fix #71

This PR changes the configuration file used by readthedocs, to use the github version of sphinx instead of the latest release (1.3.1).  This fixes the problem with the autosummary tables not displaying on readthedocs.org.

There was also a problem with the travis build failing, because of a version mismatch in the numpy used to build slycot vs. python-control.  This is also fixed, simply by installing slycot before python-control in the travis build.

There are also a few lines added to the travis script to use their new "container-based infrastru [... cut]