python-control/python-control #73: add numerator degree m to control.pade

jason-s, 2015-12-16:
Pade approximants to e^-sT with equal degrees for the numerator and denominator have some undesirable properties, in particular the behavior right at time t=0. Miklós Vajta has a [nice paper exploring the effects of using a slightly lower degree for the numberator](http://wwwhome.math.utwente.nl/~vajtam/publications/temp00-pade.pdf) and it would be helpful to add an optional argument `control.pade(T, n=1, m=None)` where if `m` is `None`, it is set equal to `n`, otherwise it provides an independent degree for the numerator. (restricted to 0 &le; m &le; n).

I am not exactly familiar with the right numerical way to construct polynomials of this type; I have done some exploration by calculating the polynomial coefficients of the Taylor series of degree `m+n` for e^-sT and then using `scipy.misc.pade` to compute an nth-degree rational approximant. This seems to work fairly well for small degrees but I suspect it has deficiencies of numerical errors.

Here's some example plots showing the issue for (m,n) = (7,11) and (m,n) = (11,11):

[image]

[image]

The equal-degree rational approximation produces lower error *after* the step begins, but has a lot of oscillation during the delay time; lower degrees in the numerator produce a "quieter" delay, at the cost of a worse approxiation after the step begins.

jason-s, 2015-12-16:
Hmm, it looks like [delay.py](https://github.com/python-control/python-control/blob/master/control/delay.py) isn't doing any fancy matrix math, just expressing the coefficients recursively.

A closed-form solution to (m,n) parameterized Pade approximants to e^A are given in  Moler + Van Loan's [Nineteen Dubious Ways to Compute the Exponential of a Matrix, Twenty-Five Years Later](http://www.cs.cornell.edu/cv/researchpdf/19ways+.pdf), we don't need the matrix math here but it does give the coefficients. If I have time to try it out, I'll submit a pull request (I'm not too optimistic about this, though).

slivingston, 2016-03-08:
Closed via pull request #76.

closed 2016-03-08 by slivingston