python-control/python-control #79: clarify license

jason-s, 2016-03-05:
Could you clarify the python-control license, either in the README file or in a top-level LICENSE.txt file? thanks!

slivingston, 2016-03-06:
It is [the BSD 3-Clause License](http://opensource.org/licenses/BSD-3-Clause) and is declared [in setup.py](https://github.com/python-control/python-control/blob/49045ca3dc138dd5ad63b0bbda921c6b7693918d/setup.py#L18). There are also copies in many of the source files, e.g., [control/__init__.py](https://github.com/python-control/python-control/blob/49045ca3dc138dd5ad63b0bbda921c6b7693918d/control/__init__.py). However, I agree that a top-level file should be added.

slivingston, 2016-03-06:
Shall the [License section of the README](https://github.com/tulip-control/tulip-control/blob/96f9c289e16e077e0f9be69c7fa8ec3f4bee08ee/README.rst#license) of [TuLiP](http://tulip-control.org) be copied here? We could also copy [the LICENSE file from there](https://github.com/tulip-control/tulip-control/blob/96f9c289e16e077e0f9be69c7fa8ec3f4bee08ee/LICENSE). Caltech is one of the copyright holders in both cases, but we may need to change some text since there are other distinct contributors.

murrayrm, 2016-03-06:
I would be too to copy the license information and the LICENSE file.  I'll try to take care of this the next time I have time to merge in some pull requests.  If someone else gets to it first, that would be great.  (Feel free to integrate a few pull requests as well -:).

murrayrm, 2016-05-30:
Addressed in commit 237d1f0db007efc8814beb39887ca0ebdf592bec

closed 2016-05-30 by murrayrm