python-control/python-control #80: Instructions in README don't install newest version of control

aramakri, 2016-03-10:
I followed the instructions on README for installing control. This installed an earlier version. I tried to use the phase_plot function but it resulted in bugs because the following line could not run in the rlocus.py file:
"from . import xferfcn"

This seemed to be an issue previously as someone else also brought this up on sourceforge: https://sourceforge.net/p/python-control/mailman/message/30029817/

To install the latest version which did not have this bug I had to fork your repository again and then run the following in terminal:
pip install https://github.com/aramakri/python-control/zipball/master

slivingston, 2016-03-10:
Could you define "newest version"? `pip install control` as [in the README](https://github.com/python-control/python-control/blob/0b26e062a6cd82da8a2420ebc3957e5267e6e116/README.rst#installation) will install [from PyPI](https://pypi.python.org/pypi/control), which has the current release. If you are referring to the tip of master branch of the repository, then note that a `git clone` command is also [given in the README](https://github.com/python-control/python-control/blob/0b26e062a6cd82da8a2420ebc3957e5267e6e116/README.rst#development).

Does this address your question? In any case, it is certainly not necessary to fork the repo.

slivingston, 2016-03-14:
@aramakri please re-open or elaborate if you think that an issue remains.

closed 2016-03-14 by slivingston