python-control/python-control #81: tf() and tf2ss() do not seem to work for static gains

MauriceRienks, 2016-03-21:
Creating a transfer function object of a static gain results in the ValueError: A must be square.

For example: tf([1.],[1.])

cwrowley, 2016-03-21:
I just verified this for `tf2ss` (using slycot)--definitely looks like a bug.  Note that there are two cases that may happen when you call `tf2ss`, depending on whether slycot is installed.  In fixing this bug, it would be good to add tests that test both versions (with and without slycot).

slivingston, 2017-02-02:
The case of static gains when Slycot is installed is now handled, given changes from <https://github.com/python-control/Slycot/pull/5>, which will be in the next release of Slycot.

slivingston, 2017-02-15:
fixed in PR #129, changes from which are now in `master` branch and will appear in the next release.

closed 2017-02-15 by slivingston