python-control/python-control #82: _remove_useless_states() can change types to ndarrays

manuelli, 2016-04-07:
It seems that if you enter the if statement [here](https://github.com/python-control/python-control/blob/master/control/statesp.py#L182-L186) then it replaces self.A with`self.A` with `zeros(1,1)` which is of type ndarray. This will cause problems later as the library is taking advantage of the fact that * is matrix multiplication not elementwise multiplcation as it would be for ndarrays. 

Here is an example of the problem

```python
sys1 = control.StateSpace(0,0,0,0)
sys2 = control.StateSpace(1,0,0,0)
print "type(sys1.A)", type(sys1.A)
print "type(sys2.A)", type(sys2.A)

type(sys1.A) <type 'numpy.ndarray'>
type(sys2.A) <class 'numpy.matrixlib.defmatrix.matrix'>
```

roryyorke, 2016-08-12:
Fixed in #107 (and it's follow-on, #108); above sample now gives

```
type(sys1.A) <class 'numpy.matrixlib.defmatrix.matrix'>
type(sys2.A) <class 'numpy.matrixlib.defmatrix.matrix'>
```

slivingston, 2016-08-14:
The link to the source code given in the opening post is floating (i.e., it tracks the tip of master branch). For posterity, a link that is fixed with respect to the current master is <https://github.com/python-control/python-control/blob/58a3ff2036aab49b63d07684c60a35d59cb395f3/control/statesp.py#L182-L186>.

roryyorke, 2017-01-01:
Bug not present in d8b07cb; most likely fixed by #110, which superseded #107 and #108.

slivingston, 2017-01-01:
Yes, I agree that it is fixed. I just ran the example from the opening post using the current tip of `master` branch (32f13bcdf32fb7d5513a5a06ba0ca3296decdf4d), and the output is
```
type(sys1.A) <class 'numpy.matrixlib.defmatrix.matrix'>
type(sys2.A) <class 'numpy.matrixlib.defmatrix.matrix'>
```
as reported earlier in <https://github.com/python-control/python-control/issues/82#issuecomment-239524227>.

closed 2017-01-01 by slivingston