python-control/python-control #83: Some way of stepping a system?

tomkcook, 2016-04-14:
I'm not sure if I'm missing something really obvious here.

I can simulate a system given a complete input time series using `control.lsim()`.

But how can I simulate a system where I get the samples one at a time and want to know the output after each one?  For instance, if I'm using it to control something or process audio or whatever?

The best I've come up with so far is to convert it to a state-space representation (if it isn't already), discretise it using the `sample()` function (again, if it isn't already) and manually run the system equations.  Is there not some way of doing this built in?

cwrowley, 2016-04-19:
I don't believe there is a ready-made function to do what you are looking for, but I am curious why you want to simulate the system one sample at a time.  One reason could be that you're determining the input from some sort of feedback law; in that case, one would normally make a new closed-loop system, for instance using `control.feedback`, and simulate that, for instance with `control.lsim`.  Another reason could be that you'd like to do something in real time, with actual hardware, and then I think you'd be better off just manually running the system equations, as you describe.

tomkcook, 2016-04-19:
I'm not working in realtime as such, but I am trying to integrate a python.control system into a larger simulation where it forms part of a feedback loop where the rest is non-python and so the samples come one at a time.

It'd be very useful to have a way of 'stepping' a continuous system like this, but I realise there's potentially a lot of work involved - everyone will have their own ideas about which integrator to use and as soon as you get into Runge-Kutta and related methods then you have to think about tolerances, steps backwards in time etc.

closed 2023-06-05 by sawyerbfuller