python-control/python-control #87: Incorrect gain margin crossover frequency

endolith, 2016-04-30:
```
L = TransferFunction([1e15], [1e1, 1.01e7, 1e11])
mag, phase, omega = bode(L, omega=2*pi*logspace(log10(10**2), log10(10**8), 500), dB=True, Hz=True)
gm, pm, sm, wg, wp, ws = stability_margins((mag, phase, omega))
```

wg should be "Gain margin crossover frequency (where phase crosses -180 degrees)".  From the graph, it should be around 100 MHz, but wg is 1.58 MHz.

wp should be "Phase margin crossover frequency (where gain crosses 0 dB)".  From the graph, this should be around 1.5 MHz, and wp is 1.498 MHz, so that looks right.

(Also, `gm, pm, sm, wg, wp, ws = stability_margins(L)` gives different results, with gm and wg == None.  I don't see anything in the docstring about why this would happen.)

endolith, 2016-05-02:
Actually it only has 2 poles, so it never actually reaches 180 degrees, so it makes sense that it would return wg == None.  So why does it return something when given `mag, phase, omega`?  It's extrapolating to guess where the phase crosses 180?

repagh, 2017-06-26:
I think with the current implementation this does not work numerically. For finding the phase and gain crossings the stability_margins function uses a "smooth" frequency response (FRD) representation, which creates a spline representation with frequency as parameter and real+complex part of the frequency response as output. 

You try to represent a range of 6 decades in frequency and 6 decades in gain (120 dB) with that, with 500 points spread out over the decades. I guess we need a different parametrisation for the smooth FRD variant for that. 

Yes, I know your question is from over a year ago, but I am writing this down as a reminder to myself too.

murrayrm, 2018-07-14:
This is working correctly in 0.8.0.

closed 2018-07-14 by murrayrm