python-control/python-control #90: can't find slycot module 'sb02md' or 'sb02nt'

Jeet1994, 2016-05-05:
I am trying to fing do the LQR using the python-control library.

However, each time I try doing it, it get the following error: 
`can't find slycot module 'sb02md' or 'sb02nt'`

My code is as follows:
```
import control
import numpy as n
a = n.matrix('1 0; 0 1')
b = n.matrix('0;1')
q = n.matrix('1 0; 0 1')
r = n.matrix('1')
K,S,E = lqr(a, b, q, r, [1])

```

The complete error message is :
```
RuntimeError: module compiled against API version a but this version of numpy is 9

Traceback (most recent call last):
  File "<pyshell#13>", line 1, in <module>
    K,S,E = lqr(a, b, q, r, [1])
  File "build\bdist.win-amd64\egg\control\statefbk.py", line 191, in lqr
    raise ControlSlycot("can't find slycot module 'sb02md' or 'sb02nt'")
ControlSlycot: can't find slycot module 'sb02md' or 'sb02nt'
```

whiplash01, 2016-05-05:
Do you have Slycot module built on your system?

Jeet1994, 2016-05-06:
I installed it using the files found at : http://www.lfd.uci.edu/~gohlke/pythonlibs/

Then, I used pip to install it.

slivingston, 2016-05-06:
To be sure that the Slycot installation is found, try
```sh
python -c 'import slycot'
```
which should return without error. If you do not have easy access to a terminal, try creating a file named tryslycot.py with contents
```python
import slycot
```
and then run tryslycot.py as you ran the code in your opening post.

slivingston, 2016-05-06:
If you are using Windows, consider instead using [Anaconda](https://www.continuum.io/why-anaconda). Note that installing Anaconda might conflict with your current arrangement of stuff from <http://www.lfd.uci.edu/%7Egohlke/pythonlibs/>, so perhaps attempt more debugging before switching.

Jeet1994, 2016-05-07:
@slivingston When I did :
`import slycot`

I got the following error:
```
RuntimeError: module compiled against API version a but this version of numpy is 9

Traceback (most recent call last):
  File "<pyshell#0>", line 1, in <module>
    import slycot
  File "G:\Packaged_Software_bundles\WinPython-64bit-2.7.10.3\python-2.7.10.amd64\lib\site-packages\slycot\__init__.py", line 18, in <module>
    from .analysis import ab01nd,ab05md,ab05nd,ab07nd,ab08nd, ab09ad, \
  File "G:\Packaged_Software_bundles\WinPython-64bit-2.7.10.3\python-2.7.10.amd64\lib\site-packages\slycot\analysis.py", line 21, in <module>
    from . import _wrapper
ImportError: numpy.core.multiarray failed to import
```

slivingston, 2016-05-07:
It appears to be some corruption, e.g., ABI incompatibility, in your NumPy installation. You can run the NumPy test suite using
```sh
python -c 'import numpy; numpy.test("full")'
```

slivingston, 2016-05-07:
[You wrote above](https://github.com/python-control/python-control/issues/90#issuecomment-217387579) that you used `pip` to install slycot. It is possible that the sources were fetched from PyPI instead of using the files provided at <http://www.lfd.uci.edu/%7Egohlke/pythonlibs/>. Can you confirm the command that you used to install from the Wheel file?

If the above fails, try Anaconda.

Jeet1994, 2016-05-07:
There was a problem with the version of Numpy I was using. I upgraded the numpy from the same place I downloaded Slycot, then did `pip install package-name.whl` and then import slycot and it worked. 
I will now try to see if the LQR works now. :+1:

Jeet1994, 2016-05-07:
I am getting a new error now. It says :
```
Traceback (most recent call last):
  File "<pyshell#6>", line 1, in <module>
    K,S,E = lqr(a, b, q, r, [1])
NameError: name 'lqr' is not defined
```

Jeet1994, 2016-05-07:
Looks like if I do `from control import *` instead of `import control `, the `K,S,E = lqr(A,B,Q,R)` works perfectly.

In addition to that, the example I took isn't controllable. I used the example used at : [Here--Page10](https://www.researchgate.net/file.PostFileLoader.html?id=568b60906307d93a818b456a&assetKey=AS%3A314419253317633%401451974799341)

And it worked like a charm.

Thanks a lot for your help. It was a pleasure talking to you. :+1:

closed 2016-05-08 by slivingston