python-control/python-control #95: Organize unit tests to follow the NumPy/SciPy guidelines

murrayrm, 2016-06-01:
Mentioned as part of PR #78: By the way, @murrayrm, it might be good at some point to organize the existing tests better, to follow the NumPy/SciPy guidelines, for instance so that tests for module foo.py are in test_foo.py. Many are already like this, but some modules do not have corresponding test modules.

slivingston, 2016-06-07:
For reference, the NumPy/SciPy Testing Guidelines are in the numpy repository at [doc/TESTS.rst.txt](https://github.com/numpy/numpy/blob/175476fec763984573e60485a447978cbcb381c0/doc/TESTS.rst.txt).

bnavigator, 2020-08-21:
The NumPy testing guidelines are now at https://numpy.org/doc/stable/reference/testing.html and replaced nose with pytest. (Nose is deprecated in general). Note that they primarily refer to NumPy/SciPy modules, which python-control is not.

With #438 this is mostly resolved.