python-control/python-control #96: Issue with control.phase_plot

aramakri, 2016-06-07:
In the python-control documentation:
http://python-control.readthedocs.io/en/latest/generated/control.phase_plot.html

It says that the phase_plot function "Produces a vector field or stream line plot for a planar system."

It also shows how to plot only streamlines:
"phase_plot(func. X0=(...), T=Tmax, ...)"

I followed this direction with this code:
"def d1(x1x2,t):
    x1,x2 = x1x2
    return np.array([x2, x2 - 2*x1])

x1x2_0 = np.array([[-1.,1.],[-1.,-1.],[1.,1.],[1.,-1.],[-1.,0.],[1.,0.],[0.,-1.],[0.,1.],[0.,0.]])

plt.figure(1)
control.phase_plot(d1,X0=x1x2_0,T=100)"

However, I get a TypeError:
"TypeError                                 Traceback (most recent call last)
<ipython-input-13-e779973f3045> in <module>()
     11 
     12 plt.figure(1)
---> 13 control.phase_plot(d1,X0=x1x2_0,T=100)
     14 
     15 def d2(x1x2,t):

/home/aramakri/cds/local/lib/python2.7/site-packages/control/phaseplot.pyc in phase_plot(odefun, X, Y, scale, X0, T, lingrid, lintime, logtime, timepts, parms, verbose)
    129         #! TODO: Add sanity checks
    130         (x1, x2) = np.meshgrid(
--> 131             frange(X[0], X[1], float(X[1]-X[0])/X[2]),
    132             frange(Y[0], Y[1], float(Y[1]-Y[0])/Y[2]));
    133 

TypeError: 'NoneType' object has no attribute '__getitem__'"

It assumes that there is a meshgrid to plot arrows and returns an error when there is none.

I am trying to plot the streamlines without the arrows because when arrows are plotted, they are so thick that the actual graph is obscured.

murrayrm, 2016-06-13:
I've verified the bug and set up a branch to fix this.

slivingston, 2016-12-26:
fixed in commit 0104e4bd036a40e45da70524a13e8356a4ad935e.

closed 2016-12-26 by slivingston