python-control/python-control #99: control.robust.hinfsyn

ochurlaud, 2016-06-27:
control version: 0.7.0
slycot: unknown version, the last from pip.

I'm not able to use the hinfsyn function.

First np (for numpy) was overwritten by an array (see my pull request).

When this is corrected, in my version of slycot, the 1st arg of `sb10ad` should not be `job` but `n`.

After correcting this, it still doesn't work:

```
In [16]: cr.hinfsyn(P,1,1)
3 1 1 1 1 1e+100
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-16-3f1d63dd2455> in <module>()
----> 1 cr.hinfsyn(P,1,1)

/home/churlaud/.local/lib/python3.4/site-packages/control/robust.py in hinfsyn(P, nmeas, ncon)
    153     mp = np.size(P.C,0)
    154     gamma = 1.e100
--> 155     out = sb10ad(n,m,mp,ncon,nmeas,gamma,P.A,P.B,P.C,P.D)
    156     gam = out[0]
    157     Ak = out[1]

/home/churlaud/.local/lib/python3.4/site-packages/slycot/synthesis.py in sb10ad(n, m, np, ncon, nmeas, gamma, A, B, C, D, job, gtol, actol, liwork, ldwork)
   1201             max(2*n*m2, m2*np2 + max(m2*m2 + 3*m2, np2*(2*np2 + m2 + max(np2, n)))))
   1202         ldwork = LW1 + max(1, LW2, LW3, LW4, LW5 + max(LW6,LW7))
-> 1203     out = _wrapper.sb10ad(job,n,m,np,ncon,nmeas,gamma,A,B,C,D,gtol,actol,liwork,ldwork)
   1204 
   1205     if out[-1] != 0:

ValueError: failed to create intent(cache|hide)|optional array-- must have defined dimensions but got (0,)
```

roryyorke, 2016-06-27:
I'm not sure what slycot you'll get from pip, but you'll need something like https://github.com/jgoppert/Slycot/pull/4 to get Sb10ad to work.  You'll also need something along the lines of https://github.com/roryyorke/python-control/tree/rory-robust-test (it looks like more than just not assigning to np); the trivial unit tests I added for hinfsyn and h2syn in that branch pass.  I did all this quite a while ago (maybe a year?), I'm afraid I can't remember the details.

Now that the slycot fix has been merged, I might as well submit a PR for this branch.

ochurlaud, 2016-06-27:
So what you mean is that I should get the master of slycot (from git) and clone python-control from your repos?

It would be great if you could give your changes. On the version I have, there are no tests for the robust libs.

ochurlaud, 2016-06-27:
I checked your PR, it seems that the tests on travis get the same error like what I'm describing above.

roryyorke, 2016-06-28:
I think the travis build is using an older slycot. The tests pass on my
ubuntu 14.04 system with up-to-date slycot from jgoppert's repo.

ochurlaud, 2016-06-28:
Error message now slightly different:
```
ValueError: failed to create intent(cache|hide)|optional array-- must have defined dimensions but got (6,0,)
```

I'm now with jgoppert/slycot:master and  roryyorke/python-control:rory-robust-test

roryyorke, 2016-06-28:
Can you post a complete example?  Does robust_test.py pass?

If I run the code from the unit test stand-alone:
```python
import control
import control.robust

p = control.ss(-1, [1, 1], [[1], [1]], [[0, 1], [1, 0]])
k, cl, gam, rcond = control.robust.hinfsyn(p, 1, 1)

print(k)
```

I get

```
A = [[-3.]]

B = [[ 1.]]

C = [[-1.]]

D = [[ 0.]]
```

ochurlaud, 2016-06-29:
The test passes.
```
A = [[ -1.17105630e+03,  -3.75569975e+05, -6.80181936e+06],
       [  1.00000000e+00,   0.00000000e+00,   0.00000000e+00],
       [  0.00000000e+00,   1.00000000e+00,   0.00000000e+00]]
B = [[1], [0], [0]]
C = [[  1.21516369e+04,   1.51505522e+06,   1.87917004e+07]]
D = [[-11.22437391]]

P = control.ss(A, B, C, D)
k, cl, gam, rcond = control.robust.hinfsyn(P, 1, 1)
```

Which outputs:
```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-12-bb60df4d05d6> in <module>()
----> 1 k, cl, gam, rcond = cr.hinfsyn(P, 1, 1)

/home/churlaud/.local/lib/python3.4/site-packages/control-dev-py3.4.egg/control/robust.py in hinfsyn(P, nmeas, ncon)
    157     np_ = np.size(P.C,0)
    158     gamma = 1.e100
--> 159     out = sb10ad(n,m,np_,ncon,nmeas,gamma,P.A,P.B,P.C,P.D)
    160     gam = out[0]
    161     Ak = out[1]

/home/churlaud/.local/lib/python3.4/site-packages/slycot/synthesis.py in sb10ad(n, m, np, ncon, nmeas, gamma, A, B, C, D, job, gtol, actol, liwork, ldwork)
   1200             max(2*n*m2, m2*np2 + max(m2*m2 + 3*m2, np2*(2*np2 + m2 + max(np2, n)))))
   1201         ldwork = LW1 + max(1, LW2, LW3, LW4, LW5 + max(LW6,LW7))
-> 1202     out = _wrapper.sb10ad(job,n,m,np,ncon,nmeas,gamma,A,B,C,D,gtol,actol,liwork,ldwork)
   1203 
   1204     if out[-1] != 0:

ValueError: failed to create intent(cache|hide)|optional array-- must have defined dimensions but got (6,0,)
```

ochurlaud, 2016-06-29:
Before you ask, same error with python 2

roryyorke, 2016-06-30:
This isn't exactly a bug in control.robust.hinfsyn, though we could do better on input checking and giving error messages.

The plant you submitted is 1x1, and you've specified 1 measurement and 1 control input; so, this plant has no "performance" inputs and outputs (these would be "w" and "z" in the formulation of https://en.wikipedia.org/wiki/H-infinity_methods_in_control_theory ).

I'm reasonably sure that each of w, z, u, and v (again, using Wikipedia's formulation) need to have at least length 1, so the smallest partitioned plant you can submit to the algorithm is 2x2 (and this is the size used in test_robust.py).

See `help(sb10ad)` for a host of other constraints on P.

I *think* all of this is discussed in Skogestad and Postlethwaite, and no doubt many other books; and I think Glover and Doyle's paper explaining the solution is available on-line.

This is all according to my slightly rusty understanding of H-infinity design; if you know better please correct me.

ochurlaud, 2016-06-30:
thx for this precisions. I think I still lack some understanding. I'm currently working with Skogestad's book about robust MIMO. I think I need to read further and that parts of theory are still not completely clear for me.

ochurlaud, 2016-08-10:
I completely forgot this issue.

As there is no tool to transform a transfer function in plant (or is it?) I used Matlab. I'm currently quite busy with my master thesis to do extra work on this, so I'll close the issue.

This was the user fault :)

closed 2016-08-10 by ochurlaud