python-control/python-control #106: Python 3.5 Support (Conda)

lgpage, 2016-08-10:
Got the following error from conda:

```
The following specifications were found to be in conflict:
  - control ==0.7.0
  - python 3.5*
Use "conda info <package>" to see the dependencies for each package.
```

roryyorke, 2016-08-10:
Have Python 3.5 packages been published?

```
$ conda search "^control$"
Fetching package metadata .........
control                      0.6.6                np19py27_6  http://conda.binstar.org/cwrowley/linux-64 
                             0.6.6                np19py33_6  http://conda.binstar.org/cwrowley/linux-64 
                             0.6.6                np19py34_6  http://conda.binstar.org/cwrowley/linux-64 
                             0.7.0                np19py27_0  http://conda.binstar.org/cwrowley/linux-64 
                             0.7.0                np19py33_0  http://conda.binstar.org/cwrowley/linux-64 
                             0.7.0                np19py34_0  http://conda.binstar.org/cwrowley/linux-64 
```

@cwrowley *has* published Slycot for Python 3.5 (for Linux,
anyway), so setting up python-control shouldn't be too difficult.

```
$ conda search "^slycot$"
Fetching package metadata .........
slycot                       0.1.1                np19py27_1  http://conda.binstar.org/cwrowley/linux-64 
                             0.1.1                np19py33_1  http://conda.binstar.org/cwrowley/linux-64 
                             0.1.1                np19py34_1  http://conda.binstar.org/cwrowley/linux-64 
                             0.2.0                np19py27_1  http://conda.binstar.org/cwrowley/linux-64 
                             0.2.0                np19py33_1  http://conda.binstar.org/cwrowley/linux-64 
                             0.2.0                np19py34_1  http://conda.binstar.org/cwrowley/linux-64 
                             0.2.0                    py26_1  http://conda.binstar.org/cwrowley/linux-64 
                          .  0.2.0                    py27_1  http://conda.binstar.org/cwrowley/linux-64 
                          .  0.2.0                    py33_1  http://conda.binstar.org/cwrowley/linux-64 
                          .  0.2.0                    py34_1  http://conda.binstar.org/cwrowley/linux-64 
                             0.2.0                    py35_1  http://conda.binstar.org/cwrowley/linux-64 
```

lgpage, 2016-08-10:
Please excuse my ignorance (I am yet to mess around with publishing conda packages), but what is the difficulty with getting windows packages for conda?

slivingston, 2016-08-11:
Building Slycot involves compiling FORTRAN. So, building `conda` packages for `control` and `slycot` on Windows involves compiling FORTRAN on Windows. It is feasible, but there might not be anyone with a build environment already prepared (and who is otherwise interested in `python-control`).

roryyorke, 2016-08-11:
Christoph Gohlke has compiled Slycot for Windows [1]; you could install this, and then install the Python control package from source (e.g., "py setup.py install" in a virtual environment).

Note the warning at the top of [1] that this version probably only works with the "official" CPython, and the specific numpy installer provided.

In the longer term, the Fortran-Python situation on Windows may improve; see [2]

[1]  http://www.lfd.uci.edu/~gohlke/pythonlibs/#slycot
[2] https://mingwpy.github.io/

KeshavA2016, 2016-09-30:
Hi all, thank you for creating and maintaining python-control. I am a recent graduate starting to learn python. I thought of starting with Anaconda distribution with python 3.5, but I see that while lapack and slycot have their corresponding py35 pkgs, I did not see a py35 specific pkg of control at http://anaconda.org/python-control/control/files . Is there anything specific that prevents python-control from working in python 3.5? Thanks in advance.

murrayrm, 2016-10-12:
No reason that python-control should not work with python 3.5.  We just need to get it set up in conda.

Juanlu001, 2017-02-13:
Perhaps you can take advantage of the conda-forge infrastructure, in case you don't know it:

http://conda-forge.github.io/

You basically submit a conda recipe and a big automation infrastructure takes care of the rest:

http://conda-forge.github.io/#add_recipe

slivingston, 2017-02-15:
@Juanlu001 thanks for the reference. Using conda-forge was proposed in #115

moorepants, 2017-06-27:
This can be closed now that control is on conda forge.

slivingston, 2017-06-27:
@lgpage please try installing from <https://anaconda.org/conda-forge/control> and re-open this issue if it fails.

closed 2017-06-27 by slivingston