python-control/python-control #111: Too few poles for MIMO tf

roryyorke, 2016-09-08:
2x2 MIMO system I/s (diagonal, integrators on diagonal) is reported as having 1 pole.  Converting to state-space gives the correct number of poles.

tested with slycot https://github.com/python-control/Slycot/commit/a1f43abe0efadd2f03115ea0c990544627205230 and python-control cdd3e73838cea0b26a7155d0b2084aa698e5eefd
```python
import numpy as np
import control

num = [ [ [1], [0] ],
        [ [0], [1] ] ]

den = [ [ [1,0], [1] ],
        [ [1],   [1,0] ] ]
          
g = control.tf(num,den)
print('g.pole()',g.pole())
print('control.ss(g).pole()',control.ss(g).pole())
```

output:
```
g.pole() [ 0.]
control.ss(g).pole() [-0. -0.]
```

murrayrm, 2017-12-30:
Took a quick look at this.  The problems seems to be in `control.TransferFunction._common_den`.  The method used to put together the common denominator for a MIMO transfer function is to create a list of all _unique_ poles that are found.  This obviously won't work in the case of repeated poles.  Definite bug.

repagh, 2018-05-30:
Does this still happen with my new implementation of _common_den ?

roryyorke, 2018-05-31:
I'll try to take a look this weekend. I see you have put the brakes on this
over at #194.

I see Matlab describe their MIMO tf pole algorithm here:
https://www.mathworks.com/help/control/ref/pole.html  (see "algorithms"
near the end)

I can't remember if that's what is intended here.

roryyorke, 2018-06-04:
Here's a moderately tricky case from Maciejowski's Multivariable Control, Example 2.2:
```python
num = [ [ [1],       [-1]      ],
        [ [1,1,-4],  [2,-1,-8] ],
        [ [1,-2],    [2,-4] ] ]

den = [ [ [1,3,2],   [1,3,2] ],
        [ [1,3,2],   [1,3,2] ],
        [ [1,1],     [1,1]   ]  ]

g = control.tf(num, den)
```

This has three poles: two at -1, and one at -2.  It also has a transmission zero at +2, which might be of interest as a test case for #205.

Matlab's pole(tf(num,den)) gives poles [-2,-2,-1,-1,-1] for this system (but pole(minreal(ss(tf(num,den)))) gives the right answer).

Octave's control toolbox version 3.0.0 resorts to minreal(ss(g)) for MIMO poles; for this example, it gives [-1,-1,-2,-2].  If I do the same thing in python-control, I get [-1,-1,-2].

ilayn, 2018-06-04:
This is system can be put into

```
[ 1/(s^2+3s+2)                                   ] * [1 -1]
[               (s^2+s-4)/(s^2+3s+2)             ]   [1  2]
[                                     (s-2)/(s+1)]   [1  1]
```

Then we have 5 poles indeed. However if we decompose to the other side by MFD then we get the four pole answer by adding `(s+2)/(s+2)` to the last row elements. But that adds a spurious zero at `-2`. 

So we shift the problem from pole to zero in terms of minimality. The actual answer is only possible to find if the common denominator can be completed to `[1, 4, 5, 2]` by taking into account that the `-1` poles have different directions but that needs a significantly more tedious search. Hence for Transfer models I think this is as good as it gets.

Since `tzero` works exclusively with state models, the answer would always be 2 or `[2, -2]` depending on which nonminimal representation is given.

roryyorke, 2018-06-05:
Is an algorithm based on extracting common denominators to left (row-wise) or right (column-wise) (or both?) a reasonable option?  Or should we just convert to state-space and get the poles that way?

ilayn, 2018-06-08:
Unfortunately, I am not aware of any algorithms that solves this satisfactorily for transfer models by obtaning Mcmillan forms. As your example nicely demonstrates, there is a need to evaluate the pole directions and I am not sure that it is worth the effort to bother with all pathological cases. 

However, the initial example clearly has a bug of not taking into account the multiplicities as Richard mentioned. I guess having more poles than the Mcmillan degree is OK since nonminimality is a common expectation. But fewer poles is not correct in my opinion and might be misleading.

murrayrm, 2018-07-02:
PR #206 claims to have fixed this problem.  It definitely fixes the problem identified by @roryyorke at the top of this issue.  For the "tricky case from Maciejowski's Multivariable Control, Example 2.2", I get the following:
```
In [4]: control.pole(g)
Out[4]: array([-2., -1., -2., -1.])

In [5]: control.pole(control.minreal(control.ss(g)))
0 states have been removed from the model
Out [5]: array([-2., -1., -1.])
```
This appears to be right, but leaving for @roryyorke to confirm and then we can close out this issue.

roryyorke, 2018-07-18:
Will try to review this weekend.

roryyorke, 2018-07-22:
Tested a007fcc, looks good to me.

closed 2018-07-22 by roryyorke