python-control/python-control #115: Add to conda-forge

jgoppert, 2016-10-03:
https://github.com/conda-forge

They have a really done a great job here, the CI builds on OSX, linux, windows, and then creates conda packages ready for install.

I just recently moved my package over: https://github.com/dronecrew/px4tools

jgoppert, 2016-10-24:
@cwrowley Want to try to tackle this with me?

cwrowley, 2016-10-25:
Sure, looks really cool.  Some issues I've run into previously when I made some conda packages:
- I never was able to figure out how to get SLICOT to link against LAPACK on OSX (I did get Linux to work though).  It's likely other people have figured that out for other packages on conda-forge though.
- I have no clue about Windows.

repagh, 2017-06-26:
Hi guys, 

This came on my list too, want to move to a python 3 installation for our aerospace Bsc control + scientific computing and programming needs. 

I have got working recipes now for both Slycot and python-control, tested for both Linux and OSX. Still have to fire up a VirtualBox Windows and check there, I started, but had to break off when it went into one of these endless update sessions.

I am using the conda-supplied compilers and libraries, so I think it should be multi-platform safe.

To be found at https://github.com/repagh/staged-recipes.git

For Slycot I am still pointing to my own version, because anaconda uses a new LAPACK library, that had some obsoleted routines that Slycot was using. Have a pull request in the pipeline for that. 

Since you raised this issue, I thought I would ask first. If you are OK with this, I will create the pull request for conda-forge.

I am currently pointing to the master for both control and Slycot, but that is not very pretty. It would be nicer to point to a real release for conda-forge. If conda-forge makes python-control more accessible to many people it would be good to define releases more often; 0.7.0 is 2 years old now, and plenty of fixes have been done since then. Who is now organising the releases?

repagh, 2017-06-26:
https://github.com/conda-forge/staged-recipes/pull/3053

Someone beat us to this, both for Slycot and control. The Slycot version circumvents the problems with newer lapack by being based on openblas.

This still means that a 2 yr old control will end up in coda. So issuing a new release now & then is still relevant.

slivingston, 2017-06-27:
@repagh tonight or tomorrow I plan to cut a new release for `control`. Regarding the general question of who organises the releases, thus far it has been ad hoc and previously @cwrowley and @murrayrm have executed the process of releasing.

moorepants, 2017-06-27:
Done: https://github.com/conda-forge/control-feedstock

slycot is currently only built for osx and windows though.

slivingston, 2017-06-27:
for posterity, here is the conda-smithy repo for slycot: https://github.com/conda-forge/slycot-feedstock

slivingston, 2017-06-27:
@moorepants thanks for creating the upstream PRs that led to the `control` and `slycot` conda-smithy repos.

@repagh thanks for also creating working recipes. Next time I will try to use this issue tracker more effectively to prevent repeated effort.

Regarding releases, I am happy to much more frequently tag bugfix releases, wherein only the micro (or "patch") number increases. As for the second number of the version triple, I think that we can increase the pace to every 6 months or less, depending on new features or desired API changes.

@jgoppert As commented by @moorepants [above](https://github.com/python-control/python-control/issues/115#issuecomment-311240719), [conda-forge builds](https://anaconda.org/conda-forge/slycot) of `slycot` for Windows are not operational yet. However, I do not interpret your opening post on this issue as requiring support for Windows in order to consider this issue resolved. It might be worth making [an issue in the tracker of Slycot](https://github.com/python-control/Slycot/issues). If I am missing something, please re-open.

closed 2017-06-27 by slivingston