python-control/python-control #120: mimo ss2tf fails without slycot

jgoppert, 2016-11-14:
Here is the transfer function:

```bash

Input 1 to output 1:
-235 s + 1.146e+04
------------------
  s^2 + 48.78 s

Input 2 to output 1:
-235 s + 1.146e+04
------------------
  s^3 + 48.78 s^2

Input 3 to output 1:
    -235 s^2 + 1.146e+04 s
------------------------------
0.008 s^3 + 1.39 s^2 + 48.78 s
```

```bash
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
/home/jgoppert/anaconda3/lib/python3.5/site-packages/control-dev-py3.5.egg/control/statesp.py in _convertToStateSpace(sys, **kw)
    646         try:
--> 647             from slycot import td04ad
    648             if len(kw):

ImportError: No module named 'slycot'

During handling of the above exception, another exception occurred:

ValueError                                Traceback (most recent call last)
<ipython-input-3-cf0438594eeb> in <module>()
      1 gains, design_data = px4tools.logsysid.control_design(data[20:80],
----> 2     do_plot=True)
      3 print(json.dumps(gains, indent=4))

/home/jgoppert/anaconda3/lib/python3.5/site-packages/px4tools-0.4.9-py3.5.egg/px4tools/logsysid.py in control_design(raw_data, do_plot, rolling_mean_window, verbose)
    372         'roll rate', roll_acc, data.ATTC_Roll,
    373         rolling_mean_window=rolling_mean_window,
--> 374         do_plot=do_plot, verbose=verbose, d_tc=d_tc)
    375 
    376     tf_integrator = control.tf((1), (1, 0))

/home/jgoppert/anaconda3/lib/python3.5/site-packages/px4tools-0.4.9-py3.5.egg/px4tools/logsysid.py in attitude_control_design(name, y, u, rolling_mean_window, do_plot, verbose, d_tc)
    330 
    331     K, G_ol_rate, G_cl_rate = pid_design(
--> 332         G_ol, K_guess, d_tc, verbose)
    333 
    334     if do_plot:
```

roryyorke, 2016-12-26:
Scipy doesn't seem to like SIMO or MISO, including when denominators are matched.  Note the error in that last case: I get the feeling `scipy.signal.lti` is intended for SISO.

```python
import sys
import numpy
import scipy
print('sys.version:',sys.version)
print('numpy.__version__:',numpy.__version__)
print('scipy.__version__:',scipy.__version__)

import numpy as np
from control.matlab import *
import scipy.signal

def testit(sys):
    print('sys is ',sys.outputs,'x',sys.inputs)
    try:
        scipy.signal.lti(np.squeeze(sys.num),np.squeeze(sys.den))
        print('success!')
    except ValueError as e:
        print('gives ValueError',e)

# 1x1
testit(tf([1],[1,1]))
# 1x1 as list-of-list-of-vectors
testit(tf([[[1]]],[[[1,1]]]))

# 1x3
testit(tf([[[-235, 1.146e4],
         [-235, 1.146E4],
         [-235, 1.146E4, 0]]],
         [[[1, 48.78, 0],
           [1, 48.78, 0, 0],
           [0.008, 1.39, 48.78]]]))

# 3x1
testit(tf([[[-235, 1.146e4]],
        [[-235, 1.146E4]],
        [[-235, 1.146E4, 0]]],
         [[[1, 48.78, 0]],
           [[1, 48.78, 0, 0]],
           [[0.008, 1.39, 48.78]]]))

# 3x1 with matched denominators
testit(tf([[[-235, 1.146e4]],
        [[-235, 1.146E4]],
        [[-235, 1.146E4, 0]]],
         [[[1, 48.78, 0]],
           [[1, 48.78, 0]],
           [[1, 48.78, 0]]]))
```

gives

```
$ python sample.py 
sys.version: 3.5.2 |Continuum Analytics, Inc.| (default, Jul  2 2016, 17:53:06) 
[GCC 4.4.7 20120313 (Red Hat 4.4.7-1)]
numpy.__version__: 1.11.1
scipy.__version__: 0.18.0
sys is  1 x 1
success!
sys is  1 x 1
success!
sys is  1 x 3
gives ValueError The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()
sys is  3 x 1
gives ValueError The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()
sys is  3 x 1
gives ValueError Denominator polynomial must be rank-1 array.
```

murrayrm, 2016-12-26:
Since I was fixing up some errors associated with not having slycot installed, I went ahead and addressed this in c90a399.  Should be fine to resolve once PR #122 is accepted.

slivingston, 2017-01-01:
addressed by commit c90a399a38babc7b496fe2f848d3ac9593b954f0, now in `master` branch from PR #122

closed 2017-01-01 by slivingston