python-control/python-control #121: Issue with slycot/_wrapper.so

TMorville, 2016-12-24:
I have some cryptic issue with slycot: 

Using the lqr in Control gives: 

```
Traceback (most recent call last):
  File "inverted_pendulum.py", line 49, in <module>
    (K, X, E) = lqr(A, B, Q, R);
  File "/Users/tobiasleedahle-morville/anaconda/lib/python3.4/site-packages/control/statefbk.py", line 191, in lqr
    raise ControlSlycot("can't find slycot module 'sb02md' or 'sb02nt'")
control.exception.ControlSlycot: can't find slycot module 'sb02md' or 'sb02nt'
```

And just using doing `import slycot` in python rises:

```
ImportError: dlopen(/Users/username/anaconda/lib/python3.4/site-packages/slycot/_wrapper.so, 2): 
Symbol not found: _dgesv_

Referenced from: /Users/username/anaconda/lib/python3.4/site-packages/slycot/_wrapper.so

Expected in: /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libBLAS.dylib
```

murrayrm, 2016-12-24:
I can confirm this error on python3.4 using conda (using ubuntu linux).  It works correctly using python3.5 (installed via conda), so appears to be something unique to the python3.4 conda build?

murrayrm, 2016-12-24:
A bit more info: the problem seems to be in the version of `numpy` that is getting loaded by `conda`.  Using `ipython`, I get the following error:

    In [1]: import slycot
    ---------------------------------------------------------------------------
    RuntimeError                              Traceback (most recent call last)
    RuntimeError: module compiled against API version 0xa but this version of numpy is 0x9
    ---------------------------------------------------------------------------
    ImportError                               Traceback (most recent call last)
    <ipython-input-1-5cdaeed6bdd3> in <module>()
    ----> 1 import slycot
    
    /home/ubuntu/miniconda3/envs/python3.4/lib/python3.4/site-packages/slycot/__init__.py in module>()
         14 
         15     # Analysis routines (6/40 wrapped)
    ---> 16     from .analysis import ab01nd,ab05md,ab05nd,ab07nd,ab08nd, ab09ad, \
         17                           ab09ax, ab09bd
        18 

    /home/ubuntu/miniconda3/envs/python3.4/lib/python3.4/site-packages/slycot/analysis.py in <module>()
         19 #       MA 02110-1301, USA.
         20 
    ---> 21 from . import _wrapper
         22 import warnings
         23 
    
    ImportError: numpy.core.multiarray failed to import

@TMorville can you give a bit more info on how you installed python-control?  Was it via conda, pip or source?

I was able to get `python-control` and `lqr` to work (tested using `examples/pvtol-lqr`) by uninstalling `control` using 

    conda uninstall control

and then installing the `python-control` package from source:

    git clone https://github.com/python-control/python-control.git
    cd python-control
    python setup.py install

This let me use the `conda` version of `slycot` (so no FORTRAN compiler required) but the latest version of `numpy` (so no issues with the version error).

TMorville, 2016-12-25:
Python is anaconda build, slycot and control was installed with pip. My PATHs are correct and which python yields correct lib. Also I've tried to build control from source as you suggested, but with the same error as I posted above. I've never seen the error related to numpy.

murrayrm, 2016-12-26:
I did a bit more checking and have reproduced this error in an anaconda build on my mac, using the `control-0.7.0` install instructions.  It looks like the issue that I ran into is that there were missing libraries that caused `slycot` to fail, which then generated the errors above.  `Slycot` was installed via pip, but looks like the required libraries didn't copy over.

@TMorville: what OS are you using?  We may need to redo the `slycot` builds on `python.org` to get pip working correctly.  (@cwrowley: will need your help with this, since I am not listed as a maintainer for `slycot`).

TMorville, 2016-12-27:
@murrayrm I'm on OS X, Sierra 10.12.2. Slycot is 0.2.0 and Control is 0.7.0. Let me know if I can do anything else.

murrayrm, 2016-12-29:
@TMorville: OK, I have a possible fix but need some help in trying it out.  First, you need to uninstall the current `slycot` installation that you have installed with pip:

    pip uninstall slycot

Now re-install slycot using conda:

    conda install -c python-control -c cyclus slycot

This command should install `slycot-0.3.0` from the python-control channel and `lapack-3.5.0` from the cyclus channel (who has compiled this standard library for osx).

Once you have done this, you should be able to `import slycot` and also run LQR using `control.lqr()`.  In particular, the file `python-control/examples/pvtol-lqr.py` from the `python-control` source distribution ran when I tested this process on my mac using a Python 3.4 anaconda environment.

Let me know if all of that works.  If so, you are good to go and I'll document this fix in the (to be created) troubleshooting section of the user manual.

TMorville, 2016-12-29:
@murrayrm implemented as you suggested. Still does not work I'm afraid. 

``` import slycot ```

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/username/anaconda/lib/python3.4/site-packages/slycot/__init__.py", line 16, in <module>
    from .analysis import ab01nd,ab05md,ab05nd,ab07nd,ab08nd, \
  File "/Users/username/anaconda/lib/python3.4/site-packages/slycot/analysis.py", line 21, in <module>
    from . import _wrapper
ImportError: dlopen(/Users/username/anaconda/lib/python3.4/site-packages/slycot/_wrapper.so, 2): Library not loaded: /opt/local/lib/libgcc/libgfortran.3.dylib
  Referenced from: /Users/username/anaconda/lib/python3.4/site-packages/slycot/_wrapper.so
  Reason: image not found
```

And trying to use ```lqr()``` yields

```
Traceback (most recent call last):
  File "/Users/username/anaconda/lib/python3.4/site-packages/control/statefbk.py", line 188, in lqr
    from slycot import sb02md
  File "/Users/username/anaconda/lib/python3.4/site-packages/slycot/__init__.py", line 16, in <module>
    from .analysis import ab01nd,ab05md,ab05nd,ab07nd,ab08nd, \
  File "/Users/username/anaconda/lib/python3.4/site-packages/slycot/analysis.py", line 21, in <module>
    from . import _wrapper
ImportError: dlopen(/Users/username/anaconda/lib/python3.4/site-packages/slycot/_wrapper.so, 2): Library not loaded: /opt/local/lib/libgcc/libgfortran.3.dylib
  Referenced from: /Users/username/anaconda/lib/python3.4/site-packages/slycot/_wrapper.so
  Reason: image not found
```

murrayrm, 2016-12-29:
Strange.  Can you try setting up a clean environment and then doing an install.  Here are the commands that I used (and that worked on my machine):

    conda create -n python3.4-slycot-test python=3.4
    source activate python3.4-slycot-test
    conda install numpy scipy matplotlib
    conda install -c python-control -c cyclus slycot control

Let me know if that generates the same error.  If so, I'll have to figure out what to uninstall from my test environment to recreate the problem.

slivingston, 2016-12-30:
The original path of `slycot` shown in the error message of the opening post is in an Anaconda directory.

@TMorville Instead of using `pip` to remove the package (as in <https://github.com/python-control/python-control/issues/121#issuecomment-269577117>), consider trying

    conda uninstall slycot

TMorville, 2016-12-30:
@murrayrm still no dice. The test env. works fine, but import slycot gives the same error as above, but now just in the test environment.

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/username/anaconda/envs/python3.4-slycot-test/lib/python3.4/site-packages/slycot/__init__.py", line 16, in <module>
    from .analysis import ab01nd,ab05md,ab05nd,ab07nd,ab08nd, \
  File "/Users/username/anaconda/envs/python3.4-slycot-test/lib/python3.4/site-packages/slycot/analysis.py", line 21, in <module>
    from . import _wrapper
ImportError: dlopen(/Users/username/anaconda/envs/python3.4-slycot-test/lib/python3.4/site-packages/slycot/_wrapper.so, 2): Library not loaded: /opt/local/lib/libgcc/libgfortran.3.dylib
  Referenced from: /Users/username/anaconda/envs/python3.4-slycot-test/lib/python3.4/site-packages/slycot/_wrapper.so
  Reason: image not found
```

@slivingston Tried `conda uninstall slycot`, and generally reproducing the entire problem without using `pip `. Also, does not seem to change anything.

repagh, 2017-06-12:
This relates to the use of a newer LAPACK, from which dgesv has been deprecated and removed. See pull request number 13 on Slycot

herovit, 2017-09-13:
Since I found this open issue when googling the same error message, it is useful here to note that the conda-forge installation process for control and slycot (see [https://github.com/python-control/python-control/issues/115](url)) solved this problem for me, presumably because it uses an older version of LAPACK.

slivingston, 2017-09-15:
thanks @herovit
note that the link in your comment has empty target, so directs back to his issue (rather #115).

murrayrm, 2017-12-30:
I think this issue is "resolved" from the point of view of python-control, in the sense that it relates to getting a working version of slycot up and running.  As @repagh noted above, the issue with lapack as been resolved in [Slycot PR #13](https://github.com/python-control/Slycot/pull/13).

closed 2017-12-30 by murrayrm