python-control/python-control #128: DC gain evaluation of State-space objects

roryyorke, 2017-01-04:
Currently (see [1]) if a state space system has eigenvalues at 0, it will return a scalar NaN for DC gain:
```python
        except LinAlgError:
            # zero eigenvalue: singular matrix
            return np.nan
```
The shape of the return is addressed in #126.

It may be possible to do better here: one relatively expensive way would be to `minreal` each input output pair and evaluate the DC gain.  This would allow finite and non-finite entries in the returned matrix, rather than all NaNs.  I'm not sure how to decide on +inf or -inf instead of NaN: in `C*inv(-A)*B`, if we find some sort of inverse of A that has infs (this is not hard using, e.g., singular value decomposition), one could still potentially add or subtract infs when multiplying by C and B, so getting NaNs.

[1] https://github.com/python-control/python-control/blob/32f13bcdf32fb7d5513a5a06ba0ca3296decdf4d/control/statesp.py#L617-L619

murrayrm, 2017-12-30:
If anyone takes up this enhancement, they should also address issue #127, which is closely related.

murrayrm, 2021-02-11:
Some related work in PR #542.