python-control/python-control #130: Convert TransferFunction's shallow copy to deep copy.

jed-frey, 2017-02-09:
I spent quite a bit of time trying to debug an error and finally stumbled on the documentation:

    # Beware: this is a shallow copy! This should be okay,
    # but be careful.

It took me a bit of digging and debugging before I realized my errors caused by the shallow copy:

    >>> print(num)
    >>> print(type(num[0]))

    [[1], [2]]
    <class 'list'>

    >>> sys1 = control.TransferFunction([num],[den])
    >>> sys1

    >>> print(num)
    >>> print(type(num[0]))

    [array([1]), array([2])]
    <class 'numpy.ndarray'>

This can make scripts behave different based on the order of operation:

    num = ...
    den = ...
    myfunc(num)
    TransferFunction(num, den)

vs.

    num = ...
    den = ...
    TransferFunction(num, den)
    myfunc(num)

To new Python users having the variables 'num' & 'den' behave different before and after calling ``TransferFunction`` can be confusing and off-putting.

roryyorke, 2017-02-09:
also reported in #112.  It probably just needs `self.num = copy.deepcopy(num)` (and the same for den) in `TransferFunction.__init__`, but I guess someone needs to read through those giant if-else-.. statements to check for complications.

murrayrm, 2017-12-30:
This has been fixed in PR #135.

closed 2017-12-30 by murrayrm