python-control/python-control #136: Dependencies under Anaconda

patrmccormack, 2017-03-04:
Folks,
Trying to install control 0.7.0 under Python 2.7 with conda gives me considerable number of downgrades as shown  below. System is Ubuntu 64 bit 1604. 
Particularly, control seems to want Numpy 1.9 whereas conda's latest build is 1.12.0
I don't think conda-forge's mismatching is the problem as I've nuked conda-forge on a different workstation and gotten the same problem.
Any ideas?
regards
pat

pat@pat-cuda:~/anaconda2$ conda install -c python-control control=0.7.0
Fetching package metadata .............
Solving package specifications: .

Package plan for installation in environment /home/pat/anaconda2:

The following NEW packages will be INSTALLED:

    control:      0.7.0-np19py27_0      python-control      
    llvm:         3.3-0                                     
    llvmpy:       0.12.7-py27_0                             
    olefile:      0.44-py27_0                               
    py2cairo:     1.10.0-py27_2                             
    system:       5.8-2                                     

The following packages will be UPDATED:

    fontconfig:   2.11.1-6              conda-forge          --> 2.11.1-6                     
    pillow:       3.4.2-py27_0          conda-forge          --> 4.0.0-py27_1                 
    qt:           4.8.7-3               conda-forge          --> 4.8.7-4                      
    tensorflow:   0.10.0rc0-np111py27_0                      --> 0.10.0-py27_0     conda-forge

The following packages will be SUPERCEDED by a higher-priority channel:

    cairo:        1.14.6-0              conda-forge          --> 1.12.18-6                    
    freetype:     2.6.3-1               conda-forge          --> 2.5.5-1                      
    matplotlib:   2.0.0rc2-np111py27_2  conda-forge/label/rc --> 1.4.3-np19py27_2             
    pango:        1.40.1-0              conda-forge          --> 1.39.0-0                     

The following packages will be DOWNGRADED due to dependency conflicts:

    astropy:      1.3-np111py27_0                            --> 1.0.4-np19py27_0             
    bottleneck:   1.1.0-np111py27_0                          --> 1.0.0-np19py27_0             
    dask:         0.13.0-py27_0                              --> 0.11.1-py27_0                
    h5py:         2.6.0-np111py27_2                          --> 2.5.0-np19py27_3             
    hdf5:         1.8.17-1                                   --> 1.8.15.1-3                   
    libpng:       1.6.27-0                                   --> 1.6.17-0                     
    numba:        0.30.0-np111py27_0                         --> 0.15.1-np19py27_0            
    numexpr:      2.6.1-np111py27_1                          --> 2.4.4-np19py27_0             
    numpy:        1.11.2-py27_0                              --> 1.9.3-py27_3                 
    pandas:       0.19.2-np111py27_0                         --> 0.17.1-np19py27_0            
    pixman:       0.34.0-0                                   --> 0.32.6-0                     
    protobuf:     3.0.0-py27_1                               --> 3.0.0b2-py27_0    conda-forge
    pyparsing:    2.1.4-py27_0                               --> 2.0.3-py27_0                 
    pytables:     3.3.0-np111py27_0                          --> 3.2.2-np19py27_0             
    scikit-image: 0.12.3-np111py27_1                         --> 0.11.3-np19py27_0            
    scikit-learn: 0.18.1-np111py27_0                         --> 0.16.1-np19py27_0            
    scipy:        0.18.1-np111py27_0                         --> 0.17.1-np19py27_1            
    statsmodels:  0.6.1-np111py27_1                          --> 0.6.1-np19py27_0

murrayrm, 2017-03-04:
Looks to me like we have a hard dependence on `numpy` 1.9 in the conda files sitting on the Anaconda cloud.  These were generated by @cwrowley quite some time ago and probably need to be updated.

@patrmccormack: Although it is not as handy, you might try installing `python-control` using pip instead of conda.  Here are the commands (from the `python-control` `README` file):

    pip install slycot   # optional
    pip install control 

If you do this within your conda environment, it should install using whatever version of numpy is installed.

patrmccormack, 2017-03-04:
That works... in 2.7 and 3.5. Thanks.

closed 2017-03-04 by patrmccormack