python-control/python-control #139: Is PID supported?

kingaza, 2017-03-25:
I am new to control system, and I just want to use PID controller in my project. 
However, I cannot find it.
Is there any example PID here, or can I implement a PID controller by this package?

murrayrm, 2017-03-25:
You can certainly use the `python-control` package to implement a PID controller.  The following code snippet will create a PID controller:

    from control import *
    kp = 1
    ki = 1
    kd = 1
    Cpid = kp + ki*tf([1], [1, 0]) + kd*tf([1, 0], [1])

kingaza, 2017-03-26:
Great! Thanks very much.
I would like to have a try.
And furthermore, is there an optimal algorithm for this PID controller?

murrayrm, 2017-03-26:
You can use any of the standard methods to design the PID controller.  A description of some standard techniques is available in Chapter 11 of the book "Feedback Systems", which you can find at http://fbsbook.org.

closed 2017-03-26 by murrayrm

kingaza, 2017-03-26:
thx again.
👍