python-control/python-control #141: Error in conversion of a MIMO transfer function model into state space

gpannocchia, 2017-03-27:
Dear python-control developers,

Thank you very much for the toolbox, which I am using in a process control course at the University of Pisa. 

I am having problems in converting a 2-input, 2-output transfer function model into state space. 
Here is the minimal code:
**************************************
import control as cnt  

# Defining a MIMO system in TF
NUM = [[[12.8], [-18.9]],[[6.6], [-19.4]]]
DEN = [[[16.7, 1.], [21.,1.]],[[10.9,1.], [14.4,1.]]]
G = cnt.TransferFunction(NUM, DEN)
print "MIMO system = ", G

# Conversion to state-space
Gss = cnt.tf2ss(G)
**************************************

Running this code (MacOS X,  python 2.7, python-control 0.70) indicates that the transfer function model is properly defined, but then the conversion function returns he following error:

**************************************
  File "/Users/gp/anaconda/lib/python2.7/site-packages/spyder/utils/site/sitecustomize.py", line 866, in runfile
    execfile(filename, namespace)

  File "/Users/gp/anaconda/lib/python2.7/site-packages/spyder/utils/site/sitecustomize.py", line 94, in execfile
    builtins.execfile(filename, *where)

  File "/Users/gp/Desktop/MIMOexample.py", line 17, in <module>
    Gss = cnt.tf2ss(G)

  File "/Users/gp/anaconda/lib/python2.7/site-packages/control/matlab.py", line 764, in tf2ss
    return _convertToStateSpace(sys)

  File "/Users/gp/anaconda/lib/python2.7/site-packages/control/statesp.py", line 667, in _convertToStateSpace
    lti_sys = lti(squeeze(sys.num), squeeze(sys.den))

  File "/Users/gp/anaconda/lib/python2.7/site-packages/scipy/signal/ltisys.py", line 761, in __init__
    self.num, self.den = normalize(*system)

  File "/Users/gp/anaconda/lib/python2.7/site-packages/scipy/signal/filter_design.py", line 1080, in normalize
    raise ValueError("Denominator polynomial must be rank-1 array.")
**************************************

Is there any error in my code or is this indeed a bug?

Thank you in advance for your time and reply.

Best regards,

   Gabriele Pannocchia

murrayrm, 2017-03-29:
This seems to be working correctly in the current master branch (__ver__ = 0.7.0.post98).  Specifically, I get the following output:

    Input 1 to output 1:
       12.8
    ----------
    16.7 s + 1
    
    Input 1 to output 2:
        6.6
    ----------
    10.9 s + 1
    
    Input 2 to output 1:
      -18.9
    --------
    21 s + 1

    Input 2 to output 2:
      -19.4
    ----------
    14.4 s + 1

Can you try grabbing the current master branch via git and see if that works?  If so, a good indication that it is time to post a new release to SourceForge.

murrayrm, 2017-03-29:
Also: can you post what version of `scipy` you are using?  It could be the problem is there (since `scipy.normalize` is throwing the error).  My tests above were done with `scipy` v0.18.1.

gpannocchia, 2017-03-29:
Dear Richard,

Thank you for the prompt reply. My version of scipy is 0.18.1. Everything is installed on a MacOS X from the anaconda distribution.

I will check out the current master branch and let you know if this solves the problem.

Best regards,

	Gabriele

gpannocchia, 2017-03-29:
I have installed the current master brach and I get the following output:

**************************
MIMO system =  
Input 1 to output 1:
   12.8
----------
16.7 s + 1

Input 1 to output 2:
    6.6
----------
10.9 s + 1

Input 2 to output 1:
  -18.9
--------
21 s + 1

Input 2 to output 2:
   -19.4
----------
14.4 s + 1

Traceback (most recent call last):

  File "<ipython-input-2-1994660fdb6d>", line 1, in <module>
    runfile('/Users/gp/Desktop/MIMOexample.py', wdir='/Users/gp/Desktop')

  File "/Users/gp/anaconda/lib/python2.7/site-packages/spyder/utils/site/sitecustomize.py", line 866, in runfile
    execfile(filename, namespace)

  File "/Users/gp/anaconda/lib/python2.7/site-packages/spyder/utils/site/sitecustomize.py", line 94, in execfile
    builtins.execfile(filename, *where)

  File "/Users/gp/Desktop/MIMOexample.py", line 17, in <module>
    Gss = cnt.tf2ss(G)

  File "build/bdist.macosx-10.7-x86_64/egg/control/statesp.py", line 1113, in tf2ss
    return _convertToStateSpace(sys)

  File "build/bdist.macosx-10.7-x86_64/egg/control/statesp.py", line 697, in _convertToStateSpace
    raise TypeError("No support for MIMO without slycot")

TypeError: No support for MIMO without slycot

**************************

So, the problem seems to be slycot missing. I will try install it and let you know if this solves the problem (also for the standard version of python-control v0.70).

Thanks,

	Gabriele

murrayrm, 2017-03-29:
For posterity, this is related to issue #120 and was fixed in commit c90a399a38babc7b496fe2f848d3ac9593b954f0.

murrayrm, 2017-03-29:
On Mar 29, 2017, at 02:35, Gabriele Pannocchia <gabriele.pannocchia@unipi.it> wrote:
    
    Do I need python3.x for slycot, right? (see below)
    
    I am using python2.7 but if necessary, I should have no problem is switching to python3.x...
    
    Gabrieles-MBP:site-packages gp$ conda install -c python-control -c cyclus slycot control
    Fetching package metadata ...............
    Solving package specifications: .
    
    
    UnsatisfiableError: The following specifications were found to be in conflict:
      - python 2.7*
      - slycot -> python 3.5*
    Use "conda info <package>" to see the dependencies for each package.
    
    Thanks,
    
    	Gabriele

There is some discussion of this in issue #136 and #106.  While not required, the issue is that we have not (yet) build slycot for the python 2.7 distribution.  Installing with pip as describe in #136 might work.

gpannocchia, 2017-03-29:
I have installed again everything (slycot and python-control) using pip with no errors:

Gabrieles-MBP:site-packages gp$ pip install slycot
The directory '/Users/gp/Library/Caches/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/Users/gp/Library/Caches/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting slycot
  Downloading slycot-0.2.0-cp27-none-macosx_10_5_x86_64.whl (1.9MB)
    100% |████████████████████████████████| 1.9MB 374kB/s 
Installing collected packages: slycot
Successfully installed slycot-0.2.0
Gabrieles-MBP:site-packages gp$ pip install control
The directory '/Users/gp/Library/Caches/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/Users/gp/Library/Caches/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting control
  Downloading control-0.7.0-py2.py3-none-any.whl (145kB)
    100% |████████████████████████████████| 153kB 2.0MB/s 
Requirement already satisfied: matplotlib in /Users/gp/anaconda/lib/python2.7/site-packages (from control)
Requirement already satisfied: numpy in /Users/gp/anaconda/lib/python2.7/site-packages (from control)
Requirement already satisfied: scipy in /Users/gp/anaconda/lib/python2.7/site-packages (from control)
Requirement already satisfied: six>=1.10 in /Users/gp/anaconda/lib/python2.7/site-packages (from matplotlib->control)
Requirement already satisfied: python-dateutil in /Users/gp/anaconda/lib/python2.7/site-packages (from matplotlib->control)
Requirement already satisfied: functools32 in /Users/gp/anaconda/lib/python2.7/site-packages (from matplotlib->control)
Requirement already satisfied: subprocess32 in /Users/gp/anaconda/lib/python2.7/site-packages (from matplotlib->control)
Requirement already satisfied: pytz in /Users/gp/anaconda/lib/python2.7/site-packages (from matplotlib->control)
Requirement already satisfied: cycler>=0.10 in /Users/gp/anaconda/lib/python2.7/site-packages (from matplotlib->control)
Requirement already satisfied: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=1.5.6 in /Users/gp/anaconda/lib/python2.7/site-packages (from matplotlib->control)
Installing collected packages: control
Successfully installed control-0.7.0

But then, I get again the first error that I posted. Any ides? Thank you.

	Gabriele

runfile('/Users/gp/Desktop/MIMOexample.py', wdir='/Users/gp/Desktop')
MIMO system =  
Input 1 to output 1:
   12.8
----------
16.7 s + 1

Input 1 to output 2:
    6.6
----------
10.9 s + 1

Input 2 to output 1:
  -18.9
--------
21 s + 1

Input 2 to output 2:
   -19.4
----------
14.4 s + 1

Traceback (most recent call last):

  File "<ipython-input-1-1994660fdb6d>", line 1, in <module>
    runfile('/Users/gp/Desktop/MIMOexample.py', wdir='/Users/gp/Desktop')

  File "/Users/gp/anaconda/lib/python2.7/site-packages/spyder/utils/site/sitecustomize.py", line 866, in runfile
    execfile(filename, namespace)

  File "/Users/gp/anaconda/lib/python2.7/site-packages/spyder/utils/site/sitecustomize.py", line 94, in execfile
    builtins.execfile(filename, *where)

  File "/Users/gp/Desktop/MIMOexample.py", line 17, in <module>
    Gss = cnt.tf2ss(G)

  File "/Users/gp/anaconda/lib/python2.7/site-packages/control/matlab.py", line 764, in tf2ss
    return _convertToStateSpace(sys)

  File "/Users/gp/anaconda/lib/python2.7/site-packages/control/statesp.py", line 667, in _convertToStateSpace
    lti_sys = lti(squeeze(sys.num), squeeze(sys.den))

  File "/Users/gp/anaconda/lib/python2.7/site-packages/scipy/signal/ltisys.py", line 761, in __init__
    self.num, self.den = normalize(*system)

  File "/Users/gp/anaconda/lib/python2.7/site-packages/scipy/signal/filter_design.py", line 1080, in normalize
    raise ValueError("Denominator polynomial must be rank-1 array.")

ValueError: Denominator polynomial must be rank-1 array.

murrayrm, 2017-03-29:
Can you confirm that slycot is installed and that python is finding it?  Something like

    import slycot
    import control
    print control.slycot_check()

should work (return True).  If that is working, then try installing the latest version of `python-control` (from GitHub) and see if that fixes the problem.   (It could be that `tf2ss` requires both `slycot` and latest version of `python-control`)

gpannocchia, 2017-03-29:
Here is what I get. As far as I can see, pip returns no errors when installing slycot but the module finds no fortran library… 

import slycot
import control
print control.slycot_check()

Traceback (most recent call last):

  File "<ipython-input-3-5e6e488a59e6>", line 1, in <module>
    import slycot

  File "/Users/gp/anaconda/lib/python2.7/site-packages/slycot/__init__.py", line 16, in <module>
    from .analysis import ab01nd,ab05md,ab05nd,ab07nd,ab08nd, ab09ad, \

  File "/Users/gp/anaconda/lib/python2.7/site-packages/slycot/analysis.py", line 21, in <module>
    from . import _wrapper

ImportError: dlopen(/Users/gp/anaconda/lib/python2.7/site-packages/slycot/_wrapper.so, 2): Library not loaded: /usr/local/lib/gcc/4.9/libgfortran.3.dylib
  Referenced from: /Users/gp/anaconda/lib/python2.7/site-packages/slycot/_wrapper.so
  Reason: image not found

murrayrm, 2017-03-29:
It looks like the problem is that `slycot` is expecting a GCC installation with `libgfortran` and it is not finding it.  Strictly speaking, this is not a `python-control` problem, but a `slycot` problem.

There are a couple of possible fixes.

* You can install `slycot` from source.  You'll need a FORTRAN compiler.  See the `README` file in the `slycot` source distribution.  It's possible that once you install FORTRAN, slycot will start working without recompiling (because the `gfortran` library might be in place.
* You can find someone who has installed `slycot` on a machine with the same architecture and try to copy over the missing library file.
* You can write a backup tf2ss function that handles MIMO and doesn't use slycot.

Sorry this is such a hassle, but one of the consequences of our choice to rely on slycot is that you have to install it and we haven't yet found a good way to make this work automatically across operating systems.

Suggestions from others welcome!

gpannocchia, 2017-03-30:
Thanks, I’ll try to make slycot work for me.

Best regards,

	Gabriele

murrayrm, 2017-12-30:
Issue tracked down to problems with `slycot` installation.  I verified that latest code generates an error message if `slycot` is not installed.  Closing.

closed 2017-12-30 by murrayrm