python-control/python-control #142: Step response output matrix not consistent between continuous-time and discrete-time

gpannocchia, 2017-04-10:
Dear Python-control developers,

When I use the step response function (control.step) on a continuous-time multivariable system the output response matrix has dimension (nsteps, noutputs), whereas the same function used on discrete-time multivariable system returns an output response matrix of dimension (noutputs, nsteps). 

Here is a sample code, which highlights the behavior. 

*********************** STARTS HERE
import control as cnt
import numpy as np

# Continuous-time system matrices
A = [[-1., 0.5],[0,-2.]]
B = np.eye(2)
C = np.eye(2)
D = np.zeros([2,2])
sysc = cnt.ss(A,B,C,D)

# Discretization
ts = 0.1
sysd = cnt.sample_system(sysc,ts)

# Step responses
tfin = 10.
Time = np.linspace(0,10,int(tfin/ts)+1)
YoutC, TimeC = cnt.step(sysc, T=Time)
YoutD, TimeD = cnt.step(sysd, T=Time)

print "Version of python-control", cnt.__version__
print "Output response matrix dimension (Continuous-Time) =", YoutC.shape
print "Output response matrix dimension (Discrete-Time) =", YoutD.shape
*********************** ENDS HERE

It returns (I am using python 2.7 under mac-osx):

Version of python-control 0.7.0
Output response matrix dimension (Continuous-Time) = (101, 2)
Output response matrix dimension (Discrete-Time) = (2, 101)

Thank you. Best regards,

	Gabriele

murrayrm, 2017-04-13:
I tried to recreate this error in the master branch and note the following differences:

- In the current version of `python-control`, the `step` function is actually `step_response`
- In the current version of `python-control`, the `step_response` function only handles SIMO systems (and warning message is generated and the second input is ignored)

The behavior above can be recreated using the `MATLAB` compatibility module, `control.matlab`, and the discrepancy in return values is as described above.

murrayrm, 2017-04-13:
Tracked down the problem: in `control/timeresp.py`, line 324, a call is made to the `scipy.signal` library for discrete time simulation.  Here is the code around that point:

    # Discrete time simulation using signal processing toolbox
    dsys = (A, B, C, D, sys.dt)
    tout, yout, xout = sp.signal.dlsim(dsys, U, T, X0)

Since `scipy.signal` uses a different convention for the way that vector valued time series are represented (see `doc/conventions.rst`), we need to transpose the output after we get it back from `scipy.signal.dlsim` using the commands

    yout = sp.transpose(yout)
    xout = sp.transpose(xout)

I'll generate a pull request to fix this and also put in some unit tests to watch for this type of bug in the future.

murrayrm, 2017-04-15:
@gpannocchia: Thank for pointing on this bug.  Fix is now pending review (PR #143) and then will become part of the master branch.

murrayrm, 2017-04-21:
Fixed in PR #143.

closed 2017-04-21 by murrayrm