python-control/python-control #144: How to cite python-control

mdclemen, 2017-04-21:
I'd like to cite my use of python-control. Does anyone know how to do this? I came across this [article](https://guides.github.com/activities/citable-code/) for assigning a DOI to a GitHub repo.

murrayrm, 2017-04-24:
Here's what the APA has to say about it:

http://blog.apastyle.org/apastyle/2015/01/how-to-cite-software-in-apa-style.html

I think the citation might look something like:

[1] Control Systems Library for Python [software] (2017).  Retrieved from http://github.com/python-control/python-control.

Juanlu001, 2017-04-24:
I recommend python-control devs to look at zenodo https://zenodo.org/ and/or The Journal of Open Source Software http://joss.theoj.org/