python-control/python-control #153: IndexError

nnn112358, 2017-07-16:
An IndexError occurs in my environment. I want advice.

-environment:
Ubuntu 14.04(x64)
Python 2.7.6

I get an error when I run this code
```
#!/usr/bin/env python
from control.matlab import *
def main():
  A_mat = [[0, 1.], [-1, -1.]]
  B_mat = [[0. ],[1.]]
  C_mat = [[1., 0.]]
  sys1tf = ss2tf(A_mat, B_mat, C_mat,0.)
  print sys1tf
if __name__ == "__main__":
  main()
```
>$ ./test1.py 
/usr/local/lib/python2.7/dist-packages/scipy/signal/filter_design.py:1452: BadCoefficients: Badly conditioned filter coefficients (numerator): the results may be meaningless
  "results may be meaningless", BadCoefficients)
Traceback (most recent call last):
  File "./controltest1.py", line 12, in <module>
    main()
  File "./controltest1.py", line 8, in main
    sys1tf = ss2tf(A_mat, B_mat, C_mat,0.0)
  File "/usr/local/lib/python2.7/dist-packages/control/matlab.py", line 687, in ss2tf
    return _convertToTransferFunction(StateSpace(*args))
  File "/usr/local/lib/python2.7/dist-packages/control/xferfcn.py", line 1131, in _convertToTransferFunction
    return TransferFunction(num, den, sys.dt)
  File "/usr/local/lib/python2.7/dist-packages/control/xferfcn.py", line 166, in __init__
    isinstance(data[i][0], (int, float, complex))):
IndexError: 0-d arrays can't be indexed

But, this code does not cause problems.

```
#!/usr/bin/env python
from control.matlab import *

def main():
  A_mat = [[0, 1.], [-1, -1.]]
  B_mat = [[0. ],[1.]]
  C_mat = [[1., 0.00001]]
  sys1tf = ss2tf(A_mat, B_mat, C_mat,0.)
  print sys1tf

if __name__ == "__main__":
  main()
```
>$ ./test1.py
> 1e-05 s + 1 / s^2 + s + 1

The difference between a problem code and a no-problem code is that C is different.
I think that no problem has happened even if I enter a similar code in MATLAB.

roryyorke, 2017-07-16:
Confirmed with scipy 0.19.0; it doesn't happen with 0.13.3.  With 0.19.0 (or earlier, don't know when this changed), it looks like the numerator is a 0-dimensional array in this case.  This diff, which is relative to the 0.7.0 tags, appears to solve the bug, but how correct it is in general I'm not sure.

```diff
diff --git a/control/xferfcn.py b/control/xferfcn.py
index 77f7bc3..2870fbd 100644
--- a/control/xferfcn.py
+++ b/control/xferfcn.py
@@ -155,7 +155,8 @@ class TransferFunction(Lti):
         # but be careful.
         data = [num, den]
         for i in range(len(data)):
-            if isinstance(data[i], (int, float, complex)):
+            if (isinstance(data[i], (int, float, complex)) or
+                (isinstance(data[i], ndarray) and data[i].shape == ())):
                 # Convert scalar to list of list of array.
                 if (isinstance(data[i], int)):
                     # Convert integers to floats at this point
```

nnn112358, 2017-07-17:
My environment is
Ubuntu 14.04 python 2.7.6 , scipy 0.13.3 , control 0.7.0

Depending on your patch, the problem has been solved.
Thank you.

roryyorke, 2017-07-30:
I didn't test carefully enough here.  I *can* reproduce this on 0.7.0 with Scipy 0.19.0, but not with master; this was fixed in 020734dc, specifically by checking for 0-dim arrays in TransferFunction.__init__:
```diff
diff --git a/control/xferfcn.py b/control/xferfcn.py
index 77f7bc3..0ea1515 100644
--- a/control/xferfcn.py
+++ b/control/xferfcn.py
@@ -155,7 +125,9 @@ class TransferFunction(Lti):
         # but be careful.
         data = [num, den]
         for i in range(len(data)):
-            if isinstance(data[i], (int, float, complex)):
+            # Check for a scalar (including 0d ndarray)
+            if (isinstance(data[i], (int, float, complex)) or
+                (isinstance(data[i], ndarray) and data[i].ndim == 0)):
```

Using scipy.signal.lti (a class) to do the SS -> TF conversion can raise a ```BadCoefficients``` warning from scipy, which doesn't happen if one uses scipy.signal.ss2tf (a function) instead.  The warning seems spurious to me; it may be useful in filter design, but not when working with LTI systems in general.

murrayrm, 2017-12-30:
I confirmed that everything still works in the latest version of the code (which has the fix that @roryyorke mentions above).

closed 2017-12-30 by murrayrm