python-control/python-control #154: pzmap() function not work, zero not plotted

elviseno, 2017-08-08:
For the same control.TransferFunction() sys, if I called with python version control.pzmap(), it reports that

> TypeError: 'module' object is not callable

If called control.matlab.pzmap() it does plot the poles but not zeros. I checked the code that when plotting zeros, facecolors is set to 'none'. 

How can I change the color back? I think it's more reasonable to set a proper color for this line. 

Thanks.

murrayrm, 2017-08-08:
Can you provide a code snippet that generates both errors?  Thanks.

elviseno, 2017-08-09:
Hi Murray,
The control package is 0.7.0. 
A simple test code used,
```python
tf = control.TransferFunction([1,1],[1,1,1])
control.pzmap(tf)
```
It reports,

> TypeError                                 Traceback (most recent call last)
<ipython-input-57-e8f98051c5bd> in <module>()
      1 tf = control.TransferFunction([1,1],[1,1,1])
----> 2 control.pzmap(tf)
      3 # control.matlab.pzmap(tf)
      4 # pole, zero = control.matlab.pzmap(tf_mindist)
      5 plt.show()
TypeError: 'module' object is not callable

control.matlab.pzmap(tf) is working fine except that zeros are not shown. 
In pzmap file,
```python
if len(zeros) > 0:
            plt.scatter(real(zeros), imag(zeros), s=50, marker='o', facecolors='none')
```
The zeros are shown are removing facecolors setting. 

It seems the 2 pzmap functions are essentially the same one. Sorry I'm not coder and can't digg too much. 

Thanks very much.

slivingston, 2017-08-09:
so, completing the example from [above](https://github.com/python-control/python-control/issues/154#issuecomment-321139677) to include necessary imports and the `show` command, we have
```python
import matplotlib.pyplot as plt
import control

tf = control.TransferFunction([1,1],[1,1,1])
control.pzmap(tf)
plt.show()
```

slivingston, 2017-08-09:
I am able to reproduce the reported behavior of missing zero markers. My hypothesis is some previous version of Matplotlib included marker edges by default, so `facecolors='none'` was sufficient to cause circle outlines. One way to achieve it now is
```python
if len(zeros) > 0:
    plt.scatter(real(zeros), imag(zeros), s=50, marker='o',
                facecolors='none', edgecolors='g')
```

slivingston, 2017-08-09:
regarding the `TypeError` exception, I was not able to produce the error using the tip of `master` branch, whereas I can reproduce the error using the previous release version 0.7.0.

Fankbai, 2017-08-16:
def pzPlot():
	h = control.tf([2, 5, 1],[1, 2, 3])
	b = pzmap(h)
	plt.zgrid(b)
	plt.show()

slivingston, 2017-08-16:
@Fankbai context?

Fankbai, 2017-08-26:
@slivingston 
def polezero():
	h = control.tf([2.27612463, -3.07696238,  2.10964293, -0.83921913],[1])
	b = pzmap(h)
	#
	plt.show()
polezero()

Sup3rGeo, 2018-02-22:
PR #193 addresses it

murrayrm, 2018-12-28:
Fixed in PR #193.

closed 2018-12-28 by murrayrm