python-control/python-control #155: how to pzplot AR model results

Fankbai, 2017-08-26:
I've posted this question on 'statsmodels' library [#3885](https://github.com/statsmodels/statsmodels/issues/3885) , but it still * does not work out.*

the matlab code work well:
```matlab
> n = ar(data, 4)
> % get the coefficiants
> A = n.a                    % as the denominator
> B = 1                        % as the nominator
> sysInput = tf(A, B)    % as the sys typycal
> pzplot(sysInput)
```
followed this idea, I try this using python-control:

```python
>> A = mdl.params
>> B= 1  
>> ipt = control.tf(A,B)
>> pzmap(ipt)
>> plt.show()
```

it seems work, but figures display:
**MATLAB**
[image]
**Python** 
[image]

Fankbai, 2017-08-26:
I've solve it using function provided  by [Christopher Felton ](https://www.dsprelated.com/showcode/244.php)  instead of control.pzmap().

but what's wrong with control.pzmap()?

roryyorke, 2020-08-02:
pzmap seems fine to me.  In  the original Matlab example, the denominator is 1, so there should be no poles.

```python
%matplotlib inline
from control import tf, pzmap
```

```python
z = tf([1, 0], [1], -1)
g1 = z / (z ** 2 - 2 * 0.5 * 0.9 * z + 0.9 ** 2)
g2 = z / (z ** 2 - 2 * 0.2 * 0.4 * z + 0.4 ** 2)
g = g1 * g2
display(g)
```

$$\frac{z^2}{z^4 - 1.06 z^3 + 1.114 z^2 - 0.2736 z + 0.1296}$$

```python
pzmap(g);
```

[image]

closed 2020-08-20 by bnavigator