python-control/python-control #156: ss2tf breaks with zero D matrix

alchemyst, 2017-09-11:
The first order system 1/(s + 1) corresponds to the system A=-1, B=1, C=1, D=0 in State Space. These matrices appear to cause an error in `ss2tf`. My preliminary testing suggests that this happens when D=0.

```
import control

A = [[-1]]
B = [[1]]
C = [[1]]
D = [[0]]

control.ss2tf(A, B, C, D)
```

Results in 

```~/anaconda3/lib/python3.6/site-packages/scipy/signal/filter_design.py:1452: BadCoefficients: Badly conditioned filter coefficients (numerator): the results may be meaningless
  "results may be meaningless", BadCoefficients)
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
<ipython-input-1-e684b1d9f8ae> in <module>()
      6 D = [[0]]
      7
----> 8 control.ss2tf(A, B, C, D)

/Users/alchemyst/anaconda3/lib/python3.6/site-packages/control/matlab.py in ss2tf(*args)
    685     if len(args) == 4 or len(args) == 5:
    686         # Assume we were given the A, B, C, D matrix and (optional) dt
--> 687         return _convertToTransferFunction(StateSpace(*args))
    688
    689     elif len(args) == 1:

/Users/alchemyst/anaconda3/lib/python3.6/site-packages/control/xferfcn.py in _convertToTransferFunction(sys, **kw)
   1129             # print(den)
   1130
-> 1131         return TransferFunction(num, den, sys.dt)
   1132
   1133     elif isinstance(sys, (int, float, complex)):

/Users/alchemyst/anaconda3/lib/python3.6/site-packages/control/xferfcn.py in __init__(self, *args)
    164                     data[i] = [[array([data[i]])]]
    165             elif (isinstance(data[i], (list, tuple, ndarray)) and
--> 166                     isinstance(data[i][0], (int, float, complex))):
    167                 # Convert array to list of list of array.
    168                 if (isinstance(data[i][0], int)):

IndexError: too many indices for array
```
I am using Anaconda on Python 3.6. I installed control using pip. I attach my full environment file

I receive similarly perplexing errors when "round-tripping" transfer functions via state space. I tried this because I was worried about the forms of my matrices:

```
def roundtrip(num, den):
    G = control.tf(num, den)
    Gss = control.ss(G)
    return control.tf(Gss)
```

Now, `roundtrip([1, 1], [2, 1])` works as expected, but `roundtrip([1], [1, 1])` gives the same error as before, and so on for higher order denominators as well: `roundtrip([1], [1, 1, 1])` also fails.

[environment.txt](https://github.com/python-control/python-control/files/1292838/environment.txt)

roryyorke, 2017-09-16:
I can reproduce this on 0.7.0, but not on HEAD.  Using git bisect, I found that this was fixed in 020734dc.

closed 2017-09-17 by alchemyst

oayan, 2017-10-10:
This issue persists. Unfortunately I have exactly the same problem. using the version 0.7.0 with python 2.7 . When the D matrix is zero, I get the same error.