python-control/python-control #157: Can't compute DLQR

Sup3rGeo, 2017-10-03:
Hello,

I want to compute a DLQR controller with the values from this matfile included:
[ufsm lqr from python.zip](https://github.com/python-control/python-control/files/1352229/ufsm.lqr.from.python.zip)

However I get a 

`Traceback (most recent call last):
  File "C:/Users/Victor/PycharmProjects/convertercontrol/routine_with_matfile.py", line 14, in <module>
    control.dare(mat["A_til"], mat["B_til"], mat["Q"], mat["R"], S=None, E=None)
  File "C:\Python27\lib\site-packages\control\mateqn.py", line 700, in dare
    X = solve_discrete_are(A, B, Qmat, Rmat)
  File "C:\Python27\lib\site-packages\scipy\linalg\_solvers.py", line 354, in solve_discrete_are
    raise ValueError('Matrix A in the algebraic Riccati equation solver '
ValueError: Matrix A in the algebraic Riccati equation solver is ill-conditioned
`

exception when using python-control, whereas MATLAB can solve and calculate the gains without any error.

I believe this is a problem with scipy riccati solver. Is there any way of getting around this?

From documentation I have seen that they use different algorithms:

MATLAB:
[1] Arnold, W.F., III and A.J. Laub, "Generalized Eigenproblem Algorithms and Software for Algebraic Riccati Equations," Proc. IEEE®, 72 (1984), pp. 1746-1754.

SCIPY:

[R127] | (1, 2, 3) P. van Dooren , “A Generalized Eigenvalue Approach For Solving Riccati Equations.”, SIAM Journal on Scientific and Statistical Computing, Vol.2(2), DOI: 10.1137/0902010

[R128] | (1, 2) A.J. Laub, “A Schur Method for Solving Algebraic Riccati Equations.”, Massachusetts Institute of Technology. Laboratory for Information and Decision Systems. LIDS-R ; 859. Available online : http://hdl.handle.net/1721.1/1301

[R129] | (1, 2) P. Benner, “Symplectic Balancing of Hamiltonian Matrices”, 2001, SIAM J. Sci. Comput., 2001, Vol.22(5), DOI: 10.1137/S1064827500367993

ilayn, 2017-10-03:
Hi @Sup3rGeo If I use the following code, I am unable to reproduce the exception 

```
from scipy.io import loadmat
from control import dare

Z = scipy.io.loadmat('ufsm lqr from python.mat')
A = Z['A_til']
B = Z['B_til']
Q = Z['Q']
R = Z['R']
dare(A, B, Q, R)
``` 

In fact the values match, 

[image]

[image]

I'm using SciPy 0.19.1 and NumPy (MKL) 1.13.1

Sup3rGeo, 2017-10-04:
@ilayn thanks very much, I updated to scipy version(0.19.1) and it worked.

closed 2017-10-04 by Sup3rGeo