python-control/python-control #159: "support for __conda_version__ has been removed as of Conda-build 3.0.Try Jinja templates instead"

kangwonlee, 2017-10-22:
Travis-CI build fails with error as follows:

`ValueError: support for __conda_version__ has been removed as of Conda-build 3.0.Try Jinja templates instead: http://conda.pydata.org/docs/building/meta-yaml.html#templating-with-jinja`

Proposed [link is here](http://conda.pydata.org/docs/building/meta-yaml.html#templating-with-jinja).

https://github.com/python-control/python-control/blob/05d6bc5257d74a05ee2615d18ab7f5f58a2ec372/.travis.yml#L43

kangwonlee, 2017-10-22:
IMHO nowadays conda version is > 4.0. Why does this happen?

kangwonlee, 2017-10-22:
There were [prior warnings](https://travis-ci.org/python-control/python-control/jobs/291176646#L758) : 

`Deprecation warning: support for __conda_version__ will be removed in Conda build 3.0.Try Jinja templates instead: http://conda.pydata.org/docs/building/meta-yaml.html#templating-with-jinja`

`Setting version from __conda_version__.txt: 0.7.0`

`Deprecation warning: support for __conda_buildnum__ will be removed in Conda build 3.0.Try Jinja templates instead: http://conda.pydata.org/docs/building/meta-yaml.html#templating-with-jinja`

kangwonlee, 2017-10-22:
Seems related to [this](https://github.com/conda/conda-build/issues/1023)

kangwonlee, 2017-10-22:
Others had similar [experience](https://github.com/rdkit/conda-rdkit/issues/53)s, too, I guess.

kangwonlee, 2017-10-22:
In some other case, some [modification](https://github.com/conda/conda-build/issues/2371) seems expected.

kangwonlee, 2017-10-23:
Are the __conda_version__.txt and __conda_buildnum__.txt files from the channel? Anything that can be done to the source repository?

anton-malakhov, 2017-11-17:
It's a pity that support `__conda_version__.txt` is removed. parsing setup.py is not an option for non-python packages

murrayrm, 2017-12-30:
This has been fixed in PR #167.

closed 2017-12-30 by murrayrm