python-control/python-control #161: bode_plot: Hz option not applied to returned frequency

FalkMusick, 2017-11-03:
Hi, 
I noticed that the Hz option in the freqplot.bode_plot() function is not applied to the returned value.
Wouldn't it be better to apply the all the options to all the returned values for consistency? 

Code Section (in freqplot.py):
```python
# Get the magnitude and phase of the system
mag_tmp, phase_tmp, omega_sys = sys.freqresp(omega)
mag = np.atleast_1d(np.squeeze(mag_tmp))
phase = np.atleast_1d(np.squeeze(phase_tmp))
 phase = unwrap(phase)
if Hz:
     omega_plot = omega_sys/(2*sp.pi)
else:
     omega_plot = omega_sys
if dB: mag = 20*sp.log10(mag)
if deg: phase = phase * 180 / sp.pi

mags.append(mag)
phases.append(phase)
omegas.append(omega_sys) # (FalkMusick) <<< Use omega_plot here!?
```
(This is my first GitHub issue report. Please tell if information is not sufficient or bad.)

Cheers
FalkMusick

murrayrm, 2017-11-05:
The original intention of the Hz option was that it would just be used for plotting, since the natural (and correct) units for frequency are rad/sec.  

Not clear to me that the return value from control.bode_plot() should actually be the frequency response (if you want that, use sys.freqresp).

Thoughts from others?

FalkMusick, 2017-11-06:
Sorry. Apparently i was referring to an older version (0.6.5, '5ac5100c8ba8e0b2b58ac54e3ad158b5e11bb7b6'). In the current GitHub version the conditionals for "Hz" and "dB" are moved into if(Plot). This makes my initial issue invalid.
There is however still code outside of if(Plot) that only needs to be evaluated if Plot is True. 

```python
# Get the magnitude and phase of the system
            mag_tmp, phase_tmp, omega_sys = sys.freqresp(omega_sys)
            mag = np.atleast_1d(np.squeeze(mag_tmp))
            phase = np.atleast_1d(np.squeeze(phase_tmp))
            phase = unwrap(phase)
            nyquistfrq_plot = None
            if Hz:
                omega_plot = omega_sys / (2. * np.pi)
                if nyquistfrq:
                    nyquistfrq_plot = nyquistfrq / (2. * np.pi)
            else:
                omega_plot = omega_sys
                if nyquistfrq:
                    nyquistfrq_plot = nyquistfrq

            mags.append(mag)
            phases.append(phase)
            omegas.append(omega_sys)
            nyquistfrqs.append(nyquistfrq)
            # Get the dimensions of the current axis, which we will divide up
            #! TODO: Not current implemented; just use subplot for now

            if (Plot):
                # Magnitude plot
                ax_mag = plt.subplot(211);
                if dB:
                    pltline = ax_mag.semilogx(omega_plot, 20 * np.log10(mag), *args, **kwargs)
                else:
                    pltline = ax_mag.loglog(omega_plot, mag, *args, **kwargs)

                if nyquistfrq_plot:
                    ax_mag.axvline(nyquistfrq_plot, color=pltline[0].get_color())
```
If i understand correctly the section
```python
            nyquistfrq_plot = None
            if Hz:
                omega_plot = omega_sys / (2. * np.pi)
                if nyquistfrq:
                    nyquistfrq_plot = nyquistfrq / (2. * np.pi)
            else:
                omega_plot = omega_sys
                if nyquistfrq:
                    nyquistfrq_plot = nyquistfrq
```
has no business being outside if(Plot).

For my purposes I'll use sys.freqresp.

Cheers
FalkMusick

murrayrm, 2017-12-27:
Agree that this code should be cleaned up a bit.  Marking as an enhancement.

murrayrm, 2019-01-05:
Fixed in PR #261

closed 2019-01-05 by murrayrm