python-control/python-control #162: Maintenance

BMaxV, 2017-11-05:
Hi everyone,

I wanted to start a little project involving control theory today and it seems this is *the* python module to use.

Can I help you clean it up a bit? Is there someone to accept pull requests? Would you even accept one that breaks backward compatibility, considering this is probably used in scientific contexts?

murrayrm, 2017-11-05:
It would be great to get some help in cleaning things up.  As you can see from the backlog of pull requests, we have a bit of a bottleneck.  But I hope to spend some time cleaning things up between now and Jan 2018.

In terms of backward compatibility: several people (including me) use this package for teaching and so backwards compatibility (eg, with Python 2.7) is useful.

BMaxV, 2017-11-06:
Ok cool, but when I said breaking backwards compatibility I didn't mean python, I meant reorganizing and renaming some bits. For example, statefbk has a "ctrb" function and your StateSpace class just takes "*args" and then you unpack that to at most five values.

Anyway, do you have another channel of communication, besides the issue tracker and mailinglist?

murrayrm, 2017-11-06:
General rules for compatibility: if an example or unit test breaks, it probably requires some discussion since people may be using that feature.  For things in the `matlab` module, we try to maintain consistency with MATLAB calling conventions.

Current channels for communication are github and the mailing list, with the issue tracker preferred.

BMaxV, 2017-11-06:
That's easy enough, I'll just keep the pull requests small and manageable.

murrayrm, 2017-12-30:
PR #171 addresses some of the docstring issues listed above.
PR #163 contains some discussion about backward compatibility.
Closing this issue since there are no further actions required.

closed 2017-12-30 by murrayrm