python-control/python-control #164: Compatibility issue with SciPy 1.0

jcwinkler, 2017-11-15:
The following snippet

```python
import control as ctrl
G = ctrl.tf(1, [1, 1])
t, y = ctrl.step_response(G)
 ```

executed under Python 3.5.4 with **SciPy 1.0**, NumPy 1.13.3 and python-control 0.7 and **SlyCot NOT installed** fails to run.

After investigation it turns out that the problem arrises in the file `statesp.py`, function `_convertToStateSpace` line 641: python-control tries to solve the problem using SlyCot. Since SlyCot is not found, it falls back to the `ImportError` handling in which the SciPy's `lti` is called:
```python
            lti_sys = lti(squeeze(sys.num), squeeze(sys.den))
            return StateSpace(lti_sys.A, lti_sys.B, lti_sys.C, lti_sys.D,
                              sys.dt)
```
In SciyPy versions PRIOR to 1.0 the lti_sys object had the attributes A, B, C, D. However, in SciPy 1.0 these attributes have been removed. Hence, the code fails to execute.

Possible workaround would be (runs using SciPy 0.19 and 1.0):
```python
            from scipy.signal import tf2ss
            A, B, C, D = tf2ss(squeeze(sys.num), squeeze(sys.den))
            return StateSpace(A, B, C, D, sys.dt)
```

What do you think? Since many Windows user do not have Slycot available by default it would be nice to get this fixed.

Jan

roryyorke, 2017-11-15:
This Scipy change is documented [in the scipy 1.0 release notes](https://docs.scipy.org/doc/scipy/reference/release.1.0.0.html#backwards-incompatible-changes) (look for "scipy.signal.lti" in the linked section).

As I recall, we've been getting deprecation warnings about this for a while, so I suppose it's hardly a surprise.

It looks like 05d6bc5 (HEAD as of this writing) still has this bug.

I can't easily test the fix: I can't get conda to find Scipy 1.0.

murrayrm, 2017-12-23:
This fix has been implemented in PR #170.  Changes were required in both `statesp.py` and `xferfcn.py`.  In addition, there were some changes in the way that scalar types showed up that required some additional changes.  See PR #170 comments for details.

closed 2018-01-02 by murrayrm