python-control/python-control #182: Deprecation warning from importlib due to \dot, etc., in docstrings

roryyorke, 2018-01-06:
See, e.g., 

  https://travis-ci.org/python-control/python-control/jobs/325460249#L2717

This particular one appears to be due to `\dot` at https://github.com/python-control/python-control/blob/7b2defe28034058eecdbde9207dc5854ab3faabd/control/statesp.py#L1056 ; the `\d` is interpreted as an (invalid) escape sequence.

The fix is probably to use a raw string.

murrayrm, 2018-01-06:
This particular line (and similar ones near it) are using a Sphinx `:math:` command (on the previous line) so that they are typeset as mathematics using sphinx (and on readthedocs).  Whatever fix is used, we should try to preserve this level of processing, so that the online documentation looks nice.  Not sure how raw strings are interpreted by Sphinx.

roryyorke, 2018-01-08:
A raw string is used in one place, namely the [long, long string at the end of `control/matlab/__init__.py`](https://github.com/python-control/python-control/blob/1801c9a4850bb82b4a3d350570440f9c4893c691/control/matlab/__init__.py#L89) --- however, I don't think that string is actually used (can't find that table on the readthedocs site).

Assuming Sphinx uses the Python interpreter to load modules, which seems likely (why write a Python parser if you don't have to?), Sphinx shouldn't care about the differences between raw and ordinary strings.

Separately: I still can't get Sphinx to work.  I can hack around the error I'm getting with this patch to numpydocs.py
```
--- numpydoc.py.orig	2017-06-20 11:30:40.000000000 +0200
+++ numpydoc.py	2018-01-08 21:34:58.110632976 +0200
@@ -115,7 +115,8 @@
         return
     doc = SphinxDocString(pydoc.getdoc(obj))
     sig = doc['Signature'] or getattr(obj, '__text_signature__', None)
-    if sig:
+    from mock import Mock
+    if sig is not None and not isinstance(sig,Mock):
         sig = re.sub(sixu("^[^(]*"), sixu(""), sig)
         return sig, sixu('')
```
so it's something to do with a Mock object.  I don't know the mock library, and couldn't figure out what is being mocked.

Even after this workaround, I don't get typeset equations (e.g., for `\dot x = A x + B u` I see that text, not the equation). I do have LaTeX installed, but perhaps there's some additional Sphinx plugin required.

murrayrm, 2018-01-15:
Fixed in PR #187 by escaping backlashes.

Created a separate issue #188.