python-control/python-control #190: testMimoW123 fails when compiled from source

akum036, 2018-01-31:
Hi,

I've installed Slycot and python-control from source and have run the provided test script:
`python setup.py test`
I get one failed test:
```
======================================================================
FAIL: testMimoW123 (control.tests.robust_test.TestAugw)
MIMO plant with all weights
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/akum036/lib/python-control/control/tests/robust_test.py", line 303, in testMimoW123
    self.siso_almost_equal(w3g[0,0], p[4,2])
  File "/home/akum036/lib/python-control/control/tests/robust_test.py", line 59, in siso_almost_equal
    raise AssertionError('systems not approx equal; max num. coeff is {}\nsys 1:\n{}\nsys 2:\n{}'.format(maxnum,g,h))
AssertionError: systems not approx equal; max num. coeff is 5.244462877690618e-08
sys 1:
A = [[-1. -2.  0.  0.]
 [-3. -4.  0.  0.]
 [ 2.  3. -4. -5.]
 [ 5.  7. -6. -7.]]

B = [[1.]
 [0.]
 [2.]
 [5.]]

C = [[23. 29. 11. 13.]]

D = [[23.]]

sys 2:
A = [[-2.  0.  0.  0.  0.  0. -2.  0.]
 [ 0. -2.  0.  0.  0.  0.  0. -2.]
 [ 0.  0. -3.  0.  0.  0.  0.  0.]
 [ 0.  0.  0. -4.  0.  0.  0.  0.]
 [ 0.  0.  0.  0. -4. -5.  2.  3.]
 [ 0.  0.  0.  0. -6. -7.  5.  7.]
 [ 0.  0.  0.  0.  0.  0. -1. -2.]
 [ 0.  0.  0.  0.  0.  0. -3. -4.]]

B = [[-2.]
 [ 0.]
 [ 3.]
 [ 0.]
 [ 2.]
 [ 5.]
 [ 1.]
 [ 0.]]

C = [[ 0.  0.  0.  0. 11. 13. 23. 29.]]

D = [[23.]]
```

The difference is fairly small (~5e-8)between the two systems, but exceeds the testing threshold (1e-8). Would anyone have ideas as to why this test is now failing, assuming it used to pass in earlier releases?

Thanks,

Avi

roryyorke, 2018-01-31:
On my system [1] the error for that test is 3e-9.  This is with Slycot and python-control both from Github; commit a41ac6 and af8d4e, respectively.

I speculate that differences in LAPACK or BLAS or some other lower-level library are the source of the difference between our results. The simplest fix is to raise the threshold; there's no particular significance in the chosen value.

If there are better ways to check for approximate equivalence of LTI systems, we could use that instead.

[1] Ubuntu 14.04 64-bit, i7-3537U, Python 3.5 from Conda

murrayrm, 2018-07-18:
@akum036 Can you try this with the new slycot (0.3.3) and python-control (0.8.0) and see if this is still an issue.

PR #320 (2019-06-30): fix testMimoW123
This fixes #190

Because the `tf(minreal(g-h))` in `siso_almost_equal` misses to remove all unobservable states in `h` (=`p[i,j]`), the transfer function has a higher order and exceeds the numerical limit on some build systems.

Ran into this while trying to build a package for OpenSUSE on their build system. With the patch, the package builds fine