python-control/python-control #191: Release v0.8.0

murrayrm, 2018-02-05:
This issue is intended to track the release of version 0.8.0 of the python-control toolbox.  All of the changes that go along with this release are now done, so I am planning to use this issue to track the steps in actually getting the release out the door (since the last time we did one was a couple of years back!).

I've put an outline of what I think the steps are for generating a release on the wiki under [release instructions](https://github.com/python-control/python-control/wiki/Release-instructions).  Will start working through those and post progress here.  (I'm going to be doing this off and on in the coming week or so => might take a while to get things out the door.  If anyone wants to help, just let me know!).

murrayrm, 2018-02-05:
Question (for those in the know): should the tag for this release be `0.8.0` or `v0.8.0`?  It seems like the latter is what I see for other packages, but not sure if that will mess up the `make_version.py` script and/or other infrastructure.

murrayrm, 2018-02-05:
Another question: how does it work to use the github [releases](https://github.com/python-control/python-control/releases) mechanism.  If I create a new release on github, does it automatically create an annotated tag?  Or should I create the tag by hand (once I figure out what to call it) and then create the release using the tag?  The [github documentation on releases](https://help.github.com/categories/releases/) is pretty sparse.

moorepants, 2018-02-05:
I typically create a tag locally for a release, push the tag to github, and then make the Github "release".

cwrowley, 2018-02-05:
As for whether to name the tags `0.8.0` or `v0.8.0`, I don't think it makes much difference, though of course it would be good to pick something and be consistent.  For instance, it looks like scikit-learn uses `0.8.0`, while scikit-image and pytorch use `v0.8.0`.  I would probably suggest sticking with 0.8.0, since that's what the last release used (long ago), and that's what `make_version.py` assumes, but it is not a strong preference.

roryyorke, 2018-03-11:
Anything still blocking 0.8.0?  Are we going to do a Slycot release first?

murrayrm, 2018-03-11:
A couple of things I think we should do:
* Release [slycot](https://github.com/python-control/Slycot) 0.3.2, containing the updated Travis CI build + updated README.  This just requires doing a couple of the merges and then rebuilding via conda-forge.
* Review PR #196 and PR #198. It would be useful to have someone do a quick check on these just to make sure I didn't miss anything.
* Create the 0.8.0 tag on `python-control`
Once that is done, we can start rolling out the changes on conda-forge, readthedocs, etc using the (proposed) [release instructions](https://github.com/python-control/python-control/wiki/Release-instructions).

repagh, 2018-06-24:
OK, I just checked some things locally. 

For the new slycot version, PR Slycot/#24 and Slycot/#25 seem to be compatible, Slycot tests with Slycot/#25 are OK with openblas. I propose using the openblas 0.3.0 from conda-forge, since that is available across the board for Mac OSX, Windows and Linux. 

Slycot/#22 probably needs a small update, we need to add a remark that for anaconda installs people should use the conda-supplied compiler, instead of installing/using external compilers. 

With that, these three PR's can be merged, 

As for python control, we have the combined Slycot/control issue with the Travis build failures. I think the combination of Slycot/#25 and #206 fixes that. That takes a leap of faith, since the checks are failing, but I think it is better to forge ahead and fix things, than to spend our time on finding a path that leads to clean Tavis builds and checks. 

What do you think?

murrayrm, 2018-06-30:
Slycot v0.3.3 is now completed, so moving on to next steps:

* Review open issues and tag any that should be resolved in v0.8.0 release
* Review open pull requests and tag any that should be integrated into v0.8.0 release
* Create the 0.8.0 tag on `python-control`.  Once that is done, we can start rolling out the changes on conda-forge, readthedocs, etc using the (proposed) release instructions.

murrayrm, 2018-07-02:
I've merged all of the PR's that I think belong in version 0.8.0, so I think this version is pretty much ready for release.  The last remaining item (I think) is to have `slycot 0.3.3` created on conda-forge.

murrayrm, 2018-07-07:
Issue #217 resolved in PR #220. Starting to work on final release.

closed 2018-07-14 by murrayrm