python-control/python-control #194: Travis CI sporadically fails in control.tests.convert_test.TestConvert

murrayrm, 2018-02-23:
Travis CI tests are failing in semi-random ways and generating the following error:

```
FAIL: testConvert (control.tests.convert_test.TestConvert)
Test state space to transfer function conversion.
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/travis/build/python-control/python-control/control/tests/convert_test.py", line 86, in testConvert
    ssTransformed = matlab.ss(tfOriginal)
  File "/home/travis/build/python-control/python-control/control/statesp.py", line 1149, in ss
    return tf2ss(sys)
  File "/home/travis/build/python-control/python-control/control/statesp.py", line 1223, in tf2ss
    return _convertToStateSpace(sys)
  File "/home/travis/build/python-control/python-control/control/statesp.py", line 779, in _convertToStateSpace
    num, den = sys._common_den()
  File "/home/travis/build/python-control/python-control/control/xferfcn.py", line 821, in _common_den
    % quad.imag.max()
AssertionError: Quadratic has a nontrivial imaginary part: 1.06581e-14
```

The error is not always present and re-running the build sometimes fixes it.  It appears the problem may be due to the way that random test cases are generated (although the seed is fixed in the unit test => not sure how we are getting different results on rebuild).

roryyorke, 2018-02-24:
Two reasons I can think of for different results given the same inputs are:
1. tests were run on different hardware, e.g., CPUs with and without AVX2,
2. and uninitialized variables.

However, neither seems all that likely.

Besides that, for the assertion to be triggered (BTW, shouldn't that be `assert all(quad.imag < 10 * eps)` be `assert all(abs(quad.imag) < 10 * eps)` ?), I think a complex conjugate pairs would have to be mismatched (that is, not exactly conjugate) -- can that happen?

If this is due to round-off, why not dodge the problem with `quad = np.array([1, -2*poles[n].real, poles[n].real**2 + poles[n].imag**2])` ?

roryyorke, 2018-05-01:
@stanmoon it would be great if we could reproduce this error.  Could you let us know your CPU, OS and version, and also Python, Numpy, Scipy, and python-control versions?  If you need help figuring any of that out, let us know.

roryyorke, 2018-05-12:
In case this was due to Scipy or Numpy updates I just retried this, but I can't reproduce it on my system, Ubuntu 18.04 64-bit running on a Intel i5-5200U.

I tried the following two setups:

1. Using system Python, Numpy, etc.:

   Python 3.6.5
   Numpy 1.13.1
   Scipy 0.19.1

This probably uses "plain old" BLAS and LAPACK for linear algebra.  Not sure if
that's relevant for this test anyway.

2. Up-to-date Conda Python 3-based system:

   Python 3.6.5
   Numpy 1.14.3
   Scipy 1.10

This had MKL installed (it was a default Conda setup).

murrayrm, 2018-07-02:
Fixed in PR #206.

closed 2018-07-02 by murrayrm