python-control/python-control #195: output of margin in wrong order

padraic00, 2018-02-24:
The wiki specifies the output of margin to be gm,pm,Wcg,Wcp. This doesn't appear to be accurate. Varying proportional gains of my system should change Wcg and *not* Wcp.

murrayrm, 2018-02-24:
Changing the gain of your system will affect both the gain margin (by exactly the inverse of the change in gain) and the phase margin (because you shift the frequency at which the gain crosses unity and it is the phase at unity gain(s) that determines the phase margin).

padraic00, 2018-02-24:
You misunderstand the issue. I'm not talking about the actual margins. I'm talking frequencies.

The order of the frequencies returned in on the wiki are in the wrong order.

murrayrm, 2018-02-24:
Thanks for the extra info.  You are right that I misunderstood the issue you were pointing out, and you are correct that the documentation for the `control.margin()` function is incorrect.

Just for posterity, here is what  is going on in a bit more detail:

* In python-control, according to the documentation for `control.margins()`, `Wcg` is the gain crossover frequency (corresponding to phase margin) and `Wcp` is the phase crossover frequency (corresponding to gain margin).  What the documentation says is the order of the return values is:
    * gm = Gain margin
    * pm = Phase margin (in degrees)
    * Wcg = Gain crossover frequency (corresponding to phase margin)
    * Wcp = Phase crossover frequency (corresponding to gain margin) (in rad/sec)

* What is actually returned is a subset of the return values in the order of the `control.stability_margins()` function, which is
    * gm = Gain margin
    * pm = Phase margin
    * sm = Stability margin, the minimum distance from the Nyquist plot to -1
    * wg = Gain margin crossover frequency (where phase crosses -180 degrees)
    * wp = Phase margin crossover frequency (where gain crosses 0 dB)
    * ws = Stability margin frequency (where Nyquist plot is closest to -1)

* In MATLAB, the relevant function call is `[Gm,Pm,Wgm,Wpm] = margin(sys)` where `Wgm` is 'the frequency where the gain margin is measured' (from MATLAB documentation) and `Wpm` is 'the frequency where the phase margin is measured'.  This is the same as `control.stability_margins()`.

So the problem is that the *documentation* for `control.margin` is incorrect.

Since I'm doing a bit of work on the code today, I'll try to post a PR to fix this soon.  @repagh and @blinkminster: I think one or both of you worked on this in the past, so if you happen to be online and I am missing something in terms of what is going on, let me know.

murrayrm, 2018-07-02:
Fixed in PR #198.

closed 2018-07-02 by murrayrm