python-control/python-control #202: Bode plot wrong when using omega keyword

physici, 2018-05-04:
I came across an error in the bode plot function. When you specify the frequency limits of the bode plot using the 'omega'-statement, the plot seems to scale linear instead of logarithmic. Hence, the plot is wrong.

To reproduce this try the following:
`sys = control.tf([1], [1,1])
control.bode(sys)
control.bode(sys, omega=[0.1, 10])`

murrayrm, 2018-05-04:
The `omega` keyword is a *list* of frequencies, not just the endpoints.  If you want to just specify the endpoints, use `omega_limits`.

Let me know if that fixes the issue.  We should make this more clear in the docstring (it says that `omega` gives a "range" of frequencies; not clear this should be a list).

murrayrm, 2018-05-05:
Ah, that's the problem.  This has been fixed and will appear in release 0.8 (which is almost done, just hasn't been officially released and pushed on on standard platforms.

In the meantime, you can grab the latest release using git from https://github.com/python-control/python-control.git

I'm changing the title of this bug to reflect the need to tweak the documentation in 0.8.0 so we don't forget that part.

physici, 2018-05-06:
I wouldn't call it a problem. Just a glitch :)

You guys do an awesome job with the control package. Saved me a lot of money and trouble, because I did not have to by Matlab.