python-control/python-control #214: Problem with step and lsim functions in control.matlab

njtheron, 2018-07-02:
After recently updating Python under a Windows 7 64 bit operating system, I get a slycot error when I try to use the step or lsim functions from control.matlab with a transfer function object.  It seems as if the program runs into this error when it tries to convert the transfer function object to a state space object.  I do not have slycot installed on my system.  Previously these commands did work, apparently without trying to invoke slycot.  The new installation has python 3.6.5, numpy 1.14.3 and control 0.7.0 .  Is this a known issue?

murrayrm, 2018-07-02:
Can you post the error message?

Also, you can try installing the current version of control-python (0.8.0), which we are hoping to release soon.  It has lots of updates and may fix your problem.

njtheron, 2018-07-02:
Dear Richard

Thanks a lot for your speedy reply!

I have updated Python only my machine at the office so I can generate the
error message only there.  It is evening now in South Africa and I am at
home.  I'll do this tomorrow and then post the error message.

I have a temporary work around for this problem, by converting the transfer
function object into a state space object with my own function, before
calling step of lsim, so it is not a very urgent matter.

Regards

Nico Theron

-- 

Prof NJ Theron

Department of Mechanical and Aeronautical Engineering

University of Pretoria

Private bag X20

Hatfield

0028 SOUTH AFRICA

Tel.: +27-12-420 3309
E-Mail: nico.theron@up.ac.za

This message and attachments are subject to a disclaimer. Please refer to
www.it.up.ac.za/documentation/governance/disclaimer/ for full details. /
Hierdie boodskap en aanhangsels is aan 'n vrywaringsklousule onderhewig.
Volledige besonderhede is by
www.it.up.ac.za/documentation/governance/disclaimer/ beskikbaar.

njtheron, 2018-07-03:
The error message is as follows (where control.matlab has been imported into Python as ct):

---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
C:\ProgramData\Anaconda3\lib\site-packages\control\statesp.py in _convertToStateSpace(sys, **kw)
    640         try:
--> 641             from slycot import td04ad
    642             if len(kw):

ModuleNotFoundError: No module named 'slycot'

During handling of the above exception, another exception occurred:

AttributeError                            Traceback (most recent call last)
<ipython-input-5-c14ecea7334d> in <module>()
      9 figure(1)
     10 tt = linspace(0.0,4.0,800)
---> 11 yy,t1 = ct.step(T1,tt)
     12 plot(t1,yy)
     13 plot([0.5,0.5],[0.0,1.0],'r')

C:\ProgramData\Anaconda3\lib\site-packages\control\matlab.py in step(sys, T, X0, input, output, **keywords)
   1326     '''
   1327     T, yout = timeresp.step_response(sys, T, X0, input, output,
-> 1328                                    transpose = True, **keywords)
   1329     return yout, T
   1330 

C:\ProgramData\Anaconda3\lib\site-packages\control\timeresp.py in step_response(sys, T, X0, input, output, transpose, **keywords)
    481     >>> T, yout = step_response(sys, T, X0)
    482     """
--> 483     sys = _get_ss_simo(sys, input, output)
    484     if T is None:
    485         if isctime(sys):

C:\ProgramData\Anaconda3\lib\site-packages\control\timeresp.py in _get_ss_simo(sys, input, output)
    407     If input is not specified, select first input and issue warning
    408     """
--> 409     sys_ss = _convertToStateSpace(sys)
    410     if sys_ss.issiso():
    411         return sys_ss

C:\ProgramData\Anaconda3\lib\site-packages\control\statesp.py in _convertToStateSpace(sys, **kw)
    666             # the squeeze
    667             lti_sys = lti(squeeze(sys.num), squeeze(sys.den))
--> 668             return StateSpace(lti_sys.A, lti_sys.B, lti_sys.C, lti_sys.D,
    669                               sys.dt)
    670 

AttributeError: 'TransferFunctionContinuous' object has no attribute 'A'

njtheron, 2018-07-03:
Sorry, I have closed the issue by accident (not yet fully understanding the system).

byusiu, 2018-09-14:
Hi, 
I also have the problem with the control.step command. It gives me an error while I ran it.
Here is the error

  File "<ipython-input-12-7235fbf92fdd>", line 1, in <module>
    control.step_response(G)

  File "C:\Users\yub\Desktop\control\timeresp.py", line 483, in step_response
    sys = _get_ss_simo(sys, input, output)

  File "C:\Users\yub\Desktop\control\timeresp.py", line 409, in _get_ss_simo
    sys_ss = _convertToStateSpace(sys)

  File "C:\Users\yub\Desktop\control\statesp.py", line 668, in _convertToStateSpace
    return StateSpace(lti_sys.A, lti_sys.B, lti_sys.C, lti_sys.D,

AttributeError: 'TransferFunctionContinuous' object has no attribute 'A'

byusiu, 2018-09-14:
Any help will be appreciated

byusiu, 2018-09-16:
Hi, 
Thanks for the quick response! I have fixed this problem by using the new control module (0.8.0 version)
I was originally using the 0.7.0 version, and the code that gives me the error is
>>G=control.tf(1,[1,2,3])
>>t,y=control.step(G)