python-control/python-control #223: matlab.step() problem

Zeitproblem, 2018-07-09:
Hi there, im trying to get the stepresponse of a transferfunction.

import numpy as np                                      
import matplotlib.pyplot as plt  
import sympy as sym  
import control as cl

from control import matlab
from sympy import Symbol, Poly               
from scipy import signal

s = Symbol('s')
G = (s+1)/(s**2+s+1)

def System(num,den):
    
    system = matlab.tf(num,den)
    return system

s1 = System([1, 1],[1, 1, 1])

def Step(system):
    plt.figure()
    yout, T = matlab.step(system)

when i call 
Step(s1)

i get the following error:

Step(s1)
Traceback (most recent call last):

  File "<ipython-input-18-89b3a88f1417>", line 1, in <module>
    Step(s1)

  File "C:/Users/ruben/Documents/Python Workspace/test.py", line 47, in Step
    yout, T = matlab.step(system)

  File "C:\Users\ruben\Anaconda3\lib\site-packages\control\matlab.py", line 1328, in step
    transpose = True, **keywords)

  File "C:\Users\ruben\Anaconda3\lib\site-packages\control\timeresp.py", line 483, in step_response
    sys = _get_ss_simo(sys, input, output)

  File "C:\Users\ruben\Anaconda3\lib\site-packages\control\timeresp.py", line 409, in _get_ss_simo
    sys_ss = _convertToStateSpace(sys)

  File "C:\Users\ruben\Anaconda3\lib\site-packages\control\statesp.py", line 668, in _convertToStateSpace
    return StateSpace(lti_sys.A, lti_sys.B, lti_sys.C, lti_sys.D,

AttributeError: 'TransferFunctionContinuous' object has no attribute 'A'

The other plots like Nyquist an Bode are working. so i do not undertand the Problem.

murrayrm, 2018-07-09:
Which version of `python-control` are you using?  The following works under v0.8.0:
```
import matplotlib.pyplot as plt  
from control import matlab

def System(num,den):
    system = matlab.tf(num,den)
    return system

s1 = System([1, 1],[1, 1, 1])

def Step(system):
    plt.figure()
    yout, T = matlab.step(system)
    plt.plot(T, yout)

Step(s1)
plt.show()
```
(Note that `matlab.step` just returns the step response; it doesn't actually plot it (unlike MATLAB).  I've flagged that as issue #224.

Zeitproblem, 2018-07-09:
Thank you! 
I think my probleem was the plotting and I didn't notice that I worked on version 0.7.0. Now everything is fine.

closed 2018-07-18 by murrayrm