python-control/python-control #226: Questions over ploting

Zeitproblem, 2018-07-09:
Now i have another question. this time about the plots. 

I have some problems editing the plots. for example:

    plt.figure()
    plt.grid(True)
    plt.title('...')
    plt.xlabel('...')
    plt.ylabel('...')

is working for:
Step Response and Nyquist. Bode and pzmp don't change. Is there some workaround? 

I also want to add an arrow(?) to the nyquistplot thet show me the direction how the phase changes. Like on the MATLAB - Nyquist-plots.

murrayrm, 2018-07-09:
The reason this doesn't work for `bode` may be that the figure consists of two separate axes (magnitude and phase) and you have to specify the axis for some of the the plot labeling commands (eg, xtitle, ytitle).  The following code will extract the axes:
```
# Get the current axes if they already exist
fig = plt.gcf()
for ax in fig.axes:
    if ax.get_label() == 'control-bode-magnitude':
        ax_mag = ax
    elif ax.get_label() == 'control-bode-phase':
        ax_phase = ax
```

Zeitproblem, 2018-07-10:
Jea but how can i change the title of the hole diagram. i tryed to do it like I already posted but it didn't worked for Bode and pzmp. now I tryed:

    matlab.pzmap(system, title = f'Pole-Zero-Map of:\n {function}')

this is working. but only for pzmp and not for Bode.

closed 2019-07-23 by Zeitproblem