python-control/python-control #235: tf2ss fails for SIMO system

lucask07, 2018-09-10:
Trying to convert a SIMO transfer function to state space fails. The error message refers to control/xferfcn.py 

>ValueError: could not broadcast input array from shape (3) into shape (2)
> /Applications/anaconda3/anaconda3/lib/python3.6/site-packages/control/xferfcn.py(819)_common_den()"

A ```ss``` from the numerator and demominator fails as well with the same error.
```tf2ss``` has worked for me with other transfer functions (i.e. 2x2 MIMO).

- On Python 3.6.3: '3.6.3 |Anaconda custom (64-bit)| (default, Oct  6 2017, 12:04:38) \n[GCC 4.2.1 Compatible Clang 4.0.1 (tags/RELEASE_401/final)]'
- Control and scipy versions are below
- slycot is installed

```
# Check availability of slycot and versions of control, scipy
import control
import scipy
import slycot
print('Slycot available = {}'.format(control.slycot_check()))
print('Control version = {}'.format(control.__version__))
print('scipy version = {}'.format(scipy.__version__))

Rs = 4e6
Rm = 1e12
Cm = 50e-12
Cin = 5e-12

# Create a numerator and denominator 
num = [[[Rm]], [[Rm*Cm*Cin*Rs, (Rm*Cm + Cin*Rm + Cin*Rs), 1]]]
den = [[[Rm*Rs*Cm, Rs + Rm]], [[Rs*Rm*Cm, Rm + Rs]]]

# create a transfer function 
sys = control.tf(num, den)
print(sys)
# convert the TF to state-space (this fails)
sys_ss = control.tf2ss(sys) # <--- fails 
```

The output and error message is below:

> Slycot available = True
> Control version = 0.8.0
> scipy version = 0.19.1
>
> Input 1 to output 1:
>      1e+12
> /---------------
> 2e+08 s + 1e+12
> 
> Input 1 to output 2:
> 0.001 s^2 + 55 s + 1
> /--------------------
>   2e+08 s + 1e+12
> 
> ---------------------------------------------------------------------------
> ValueError                                Traceback (most recent call last)
> <ipython-input-65-53ef620c44f1> in <module>()
>      12 sys_cell = control.tf(num, den)
>      13 print(sys_cell)
> ---> 14 sys_cell_ss = control.tf2ss(sys_cell)
> 
> /Applications/anaconda3/anaconda3/lib/python3.6/site-packages/control/statesp.py in tf2ss(*args)
>    1243             raise TypeError("tf2ss(sys): sys must be a TransferFunction \
>    1244 object.")
> -> 1245         return _convertToStateSpace(sys)
>    1246     else:
>    1247         raise ValueError("Needs 1 or 2 arguments; received %i." % len(args))
> 
> /Applications/anaconda3/anaconda3/lib/python3.6/site-packages/control/statesp.py in _convertToStateSpace(sys, **kw)
>     803             # function matrix has a common denominator.
>     804             # matrices are also sized/padded to fit td04ad
> --> 805             num, den, denorder = sys.minreal()._common_den()
>     806 
>     807             # transfer function to state space conversion now should work!
> 
> /Applications/anaconda3/anaconda3/lib/python3.6/site-packages/control/xferfcn.py in _common_den(self, imag_tol)
>     817                     #print(j,i,m,len(numpoly),len(poles[j]))
>     818                     if m < 0:
> --> 819                         num[i,j,::-1] = numpoly
>     820                     else:
>     821                         num[i,j,:m:-1] = numpoly
> 
> ValueError: could not broadcast input array from shape (3) into shape (2)
> > /Applications/anaconda3/anaconda3/lib/python3.6/site-packages/control/xferfcn.py(819)_common_den()
>     817                     #print(j,i,m,len(numpoly),len(poles[j]))
>     818                     if m < 0:
> --> 819                         num[i,j,::-1] = numpoly
>     820                     else:
>     821                         num[i,j,:m:-1] = numpoly
>

ilayn, 2018-09-10:
You have a noncausal system in the second entry. Hence probably it is not supported. In matlab you probably also get a descriptor system with an `E` matrix.

lucask07, 2018-09-10:
OK, closing. Thank you for your help. I confirmed that by adding an s^2 term to the denominator of the second entry the state-space model was created without error.

closed 2018-09-10 by lucask07